kpreid/shinysdr #15: gr buffer allocation error.

corpr8, 2015-01-21:
Just installed shinysdr, using a nooElec dongle and the server fires up fine however, when the first client connects, it traps as follows:

INFO:shinysdr:127.0.0.1 - - [21/Jan/2015:11:11:41 +0000] "GET /client/plugins/shinysdr.plugins.hamlib/hamlib.js HTTP/1.1" 200 1698 "http://localhost:8100/KLyBsKjF44TGC6U7y1XGlA/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Ubuntu Chromium/39.0.2171.65 Chrome/39.0.2171.65 Safari/537.36"
INFO:shinysdr:Starting RFC 6455 conversation
INFO:shinysdr:Stream connection to  /KLyBsKjF44TGC6U7y1XGlA/radio
INFO:shinysdr:Starting RFC 6455 conversation
INFO:shinysdr:Stream connection to  /KLyBsKjF44TGC6U7y1XGlA/audio?rate=44100
INFO:shinysdr:Flow graph: Rebuilding connections
INFO:shinysdr:Flow graph: ...done reconnecting.
gr::vmcircbuf_sysv_shm: shmget(1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget(1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget(1): Invalid argument
gr::buffer::allocate_buffer: failed to allocate buffer of size 37500 KB
gr::vmcircbuf_sysv_shm: shmget(1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget(1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget(1): Invalid argument
gr::buffer::allocate_buffer: failed to allocate buffer of size 37500 KB
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted

Any thoughts greatly appreciated

kpreid, 2015-01-21:
Note this is purely a GNU Radio problem — ShinySDR doesn't use that layer directly.

A quick web search suggests that if you're on Linux you may be able to raise the system shared memory limit: http://lists.gnu.org/archive/html/discuss-gnuradio/2014-10/msg00385.html

corpr8, 2015-01-21:
Thanks for your help - that's fixed it :-)

closed 2015-01-21 by corpr8

hiteshiiti, 2015-09-12:
i am getting the same error.
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::buffer::allocate_buffer: failed to allocate buffer of size 1536000 KB
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::buffer::allocate_buffer: failed to allocate buffer of size 1536000 KB
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

the buffer size is around 1.5GB. i tried to set kenel.shmmax to 2GB but its not working. can anyone tell me what should i do.

kpreid, 2015-09-13:
@hk123 What type of hardware are you're using? What's its sample rate? Please try reducing the sample rate to see if it reduces the buffer size and lets things work. See issue #34 for details.

hk123, 2015-09-13:
i am not using any hardware. first i am trying to test my application without any hardware.

kpreid, 2015-09-13:
@hk123 I just pushed a change that might help. Please give it a try and tell me whether it makes a difference.

hk123, 2015-09-13:
@kpreid sorry i didn't tell you that i am not using shiny sdr. the application which i am trying to implement is different but the error matches with the above. can you suggest something what should i do?

kpreid, 2015-09-13:
@hk123 This is not an appropriate place for that discussion. You can try [the discuss-gnuradio mailing list](https://lists.gnu.org/mailman/listinfo/discuss-gnuradio). (I'd suggest first reducing your program to a small example that demonstrates the problem, even if it doesn't accomplish anything useful, and including that in your message.)

hk123, 2015-09-13:
@kpreid i also looked in your above described mailing list http://lists.gnu.org/archive/html/discuss-gnuradio/2014-10/msg00385.html but didn't solved my problem. anyway i will discuss it there. thanks for your help.