kpreid/shinysdr #16: many errors but final one is :exceptions.AttributeError: 'module' object has no attribute 'PY2'

vo1pwf, 2015-01-27:
out put from ""shinysdr  ~/.shinysdr_settings"" as follows

INFO:shinysdr:Log opened.
linux; GNU C++ version 4.8.1; Boost_105300; UHD_003.008.001-105-g91ae742f

gr-osmosdr v0.1.4-9-g48045b59 (0.1.5git) gnuradio 3.7.6.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR
Found Fitipower FC0013 tuner
INFO: Audio source arch: alsa
Using Volk machine: avx_64_mmx_orc
ERROR:shinysdr:Unhandled Error
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/simulate.py", line 99, in add_modulator
    mode_def = lookup_mode(mode)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/modes.py", line 144, in lookup_mode
    for mode_def in get_modes():
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/modes.py", line 136, in get_modes
    return [p for p in getPlugins(_IModeDef, plugins) if p.available]
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/plugin.py", line 209, in getPlugins
    allDropins = getCache(package)
--- <exception caught here> ---
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/plugin.py", line 167, in getCache
    provider = pluginModule.load()
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/python/modules.py", line 383, in load
    return self.pathEntry.pythonPath.moduleLoader(self.name)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/python/reflect.py", line 303, in namedAny
    topLevelPackage = _importAndCheckStack(trialname)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/python/reflect.py", line 250, in _importAndCheckStack
    reraise(excValue, excTraceback)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/rtty/__init__.py", line 35, in <module>
    import rtty  # gr-rtty
exceptions.ImportError: No module named rtty

ERROR:shinysdr:Unhandled Error
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/simulate.py", line 99, in add_modulator
    mode_def = lookup_mode(mode)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/modes.py", line 144, in lookup_mode
    for mode_def in get_modes():
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/modes.py", line 136, in get_modes
    return [p for p in getPlugins(_IModeDef, plugins) if p.available]
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/plugin.py", line 209, in getPlugins
    allDropins = getCache(package)
--- <exception caught here> ---
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/plugin.py", line 167, in getCache
    provider = pluginModule.load()
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/python/modules.py", line 383, in load
    return self.pathEntry.pythonPath.moduleLoader(self.name)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/python/reflect.py", line 303, in namedAny
    topLevelPackage = _importAndCheckStack(trialname)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/python/reflect.py", line 250, in _importAndCheckStack
    reraise(excValue, excTraceback)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/rtty/__init__.py", line 35, in <module>
    import rtty  # gr-rtty
exceptions.ImportError: No module named rtty

ERROR:shinysdr:Unhandled Error
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/simulate.py", line 99, in add_modulator
    mode_def = lookup_mode(mode)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/modes.py", line 144, in lookup_mode
    for mode_def in get_modes():
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/modes.py", line 136, in get_modes
    return [p for p in getPlugins(_IModeDef, plugins) if p.available]
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/plugin.py", line 209, in getPlugins
    allDropins = getCache(package)
--- <exception caught here> ---
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/plugin.py", line 167, in getCache
    provider = pluginModule.load()
[... 779 lines cut ...]
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/python/context.py", line 81, in callWithContext
    return func(*args,**kw)
--- <exception caught here> ---
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/internet/posixbase.py", line 614, in _doReadOrWrite
    why = selectable.doRead()
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/internet/tcp.py", line 214, in doRead
    return self._dataReceived(data)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-14.0.2-py2.7-linux-x86_64.egg/twisted/internet/tcp.py", line 220, in _dataReceived
    rval = self.protocol.dataReceived(data)
  File "/usr/local/lib/python2.7/dist-packages/txWS-0.9.1-py2.7.egg/txws.py", line 629, in dataReceived
    self.parseFrames()
  File "/usr/local/lib/python2.7/dist-packages/txWS-0.9.1-py2.7.egg/txws.py", line 456, in parseFrames
    frames, self.buf = parser(self.buf)
  File "/usr/local/lib/python2.7/dist-packages/txWS-0.9.1-py2.7.egg/txws.py", line 271, in parse_hybi07_frames
    if six.PY2:
exceptions.AttributeError: 'module' object has no attribute 'PY2'

^CINFO:shinysdr:Received SIGINT, shutting down.
INFO:shinysdr:(TCP Port 8100 Closed)
INFO:shinysdr:Stopping factory <twisted.web.server.Site instance at 0x61173f8>
INFO:shinysdr:(TCP Port 8101 Closed)
INFO:shinysdr:Stopping factory <shinysdr.web.OurStreamFactory instance at 0x5ee4248>
INFO:shinysdr:Stopping factory <txws.WebSocketFactory instance at 0x5ee4290>
INFO:shinysdr:Main loop terminated.

kpreid, 2015-01-27:
The errors earlier about module `rtty` are spurious in that the module is supposed to be optional; I've pushed a fix. Thanks for bringing it to my attention.

The error about `six.PY2` means that you have a too-old version of the module `six`. You'll have to install a more recent version.

vo1pwf, 2015-01-27:
ok....Ill try and update six......Im running base linux mint 16 if that
helps you or anyone else in the future if they receive the same error......

I did get the website up....but it sticks on loading....thats when I looked
for errors...

will report back with any new developements

vo1pwf, 2015-01-27:
built six from scratch .... started up shinysdr.... WOW WOW WOW AMAZING
great work guys n gals just absolutly amazing
WOW!

closed 2015-01-27 by vo1pwf

ch0mik, 2015-01-31:
Hello - i build the python six from sources - its Working ;)