kpreid/shinysdr #21: fetch-js-deps.sh  not placing require.js and openlayers in webstatic dir

ryanturner, 2015-02-03:
Fresh install using fetch-js-deps.sh, getting 404 for:
GET http://myhost:8000/client/require.js 
GET http://myhost:8000/client/openlayers/OpenLayers.debug.js 

Used install.py

Despite putting the files in every place I can find
/home/ryan_turner/shinysdr/build/lib.linux-x86_64-2.7/shinysdr/webstatic/client/
/home/ryan_turner/shinysdr/shinysdr/webstatic/client/
/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/webstatic/client/

The file shows up in a directory listing when I go to http://myhost:8000/client/, but when I open the file is 404s.

ryanturner, 2015-02-03:
If I execute the server using "python -m shinysdr.main" instead of "shinysdr", it works properly.

kpreid, 2015-02-03:
It's not expected that the deps files should be actually in `webstatic/client/`; they're grafted in at runtime by `shinysdr.web.WebService`.

Did you run `fetch-js-deps.sh` *before* `setup.py install`? If you did the opposite then that would explain part of it. Check whether the files exist in `/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/deps/`, and if they don't, reinstall.

ryanturner, 2015-02-03:
No, I ran setup.py install, then fetch-js-deps.sh; the readme states "Install the following software ... If you use setup.py, this will be automatic ... Then either run the script fetch-js-deps.sh"

So, now I re-read it and understand what you meant, but reading from top to bottom, I thought "Oh I should just go ahead and use setup.py to satisfy these dependecies, then Ill run fetch-js-deps.sh"

Having tons of fun with shindysdr so far, btw. Have it installed on a VM at one of our org's sites with an omni ~ 60m AGL. Currently battling a terrible connection, the uplink to that site is obscured by trees and limited to about 100 kB/s atm. Looking forward to having this resolved and then having a public shinysdr running!

kpreid, 2015-02-04:
Thanks for explaining the problem with the README. It's a bit hard to neatly explain the differences between the "use setup.py" path and the "run from this directory" path, and it could be improved. I'll leave this issue open to remind myself to do something about that case.

kpreid, 2015-02-16:
I've reorganized the installation section and added a specific sequence of steps more explanation about using `setup.py`. Hope this helps.

closed 2015-02-16 by kpreid