kpreid/shinysdr #25: how to simplify webconsole ?

ch0mik, 2015-02-08:
Hello,

I want to install it on my territorial division (OT15, http://ot15.pgk.net.pl/ ) WebSDR. I testing the software from Pieter-Tjerk, PA3FWM. Its have some setting in server-side (eg. Gain, center freq and etc). I want to do with Urs Software - how ?

PS. my test instance of ShinySDR : http://sq7mru.blogspot.com/2015/01/websdr-z-dongla-rtl-sdr-bazujac-na.html

Vy 73! de SQ7MRU, Paweł

kpreid, 2015-02-08:
If you modify some classes in `osmosdr.py`, you can remove these controls:

* In `_OsmoSDRTuning` you can change the `ctor` of `__vfo_cell` to a single-point range like `Range([(freq, freq)])`.
* In `OsmoSDRRXDriver` if you delete the methods `get_gain` and `set_gain` and the line `callback(BlockCell(self, 'gains'))` this will remove the ability (and the UI) to change the gain settings.

In the long term, I have plans to make ShinySDR more suitable for multi-user access; this will likely include a mode where most users don't have these controls (only one user has control of tuning), which will allow for the result you want without any code changes.

ch0mik, 2015-02-09:
ThX For Reply.

Regards
Pawel

closed 2015-02-09 by kpreid