kpreid/shinysdr #26: Raspberry Pi2: Missing alsa source

corpr8, 2015-02-23:
Running the latest rasbian and installing gnuradio from source, I get the following error after installing and then trying to run shinysdr:

pi@raspberrypi ~/shinysdr $ shinysdr test
INFO:shinysdr:Log opened.
linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown

gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
gr::log :INFO: audio source - Audio source arch: alsa
gr::log :ERROR: audio_alsa_source0 - [default]: No such file or directory
ERROR:shinysdr:main function encountered error
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 52, in main
    react(go)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-15.0.0-py2.7-linux-armv7l.egg/twisted/internet/task.py", line 875, in react
    finished = main(_reactor, *argv)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 47, in go
    return _main_async(reactor, argv, _abort_for_test)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-15.0.0-py2.7-linux-armv7l.egg/twisted/internet/defer.py", line 1253, in unwindGenerator
    return _inlineCallbacks(None, gen, Deferred())
--- <exception caught here> ---
  File "/usr/local/lib/python2.7/dist-packages/Twisted-15.0.0-py2.7-linux-armv7l.egg/twisted/internet/defer.py", line 1107, in _inlineCallbacks
    result = g.send(result)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 91, in _main_async
    execute_config(configObj, args.configFile)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/config.py", line 211, in execute_config
    execfile(config_file, env)
  File "test", line 15, in <module>
    config.devices.add(u'audio', shinysdr.devices.AudioDevice(''))
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/devices.py", line 335, in AudioDevice
    quadrature_as_stereo=quadrature_as_stereo)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/devices.py", line 388, in __init__
    ok_to_block=True)
  File "/usr/lib/python2.7/dist-packages/gnuradio/audio/audio_swig.py", line 133, in make
    return _audio_swig.source_make(*args, **kwargs)
exceptions.RuntimeError: audio_alsa_source

Any help greatfully appreciated (As you can see, I got a bigger workhorse to run your code!)

kpreid, 2015-02-24:
The actual error comes from gnuradio's audio source subsystem trying to find a Linux audio input device. I'm afraid I don't know much about Linux audio or ALSA in particular.

However, if you have no use for the audio input (i.e. you're not using a soundcard radio) you can just remove AudioDevice from your ShinySDR config file to skip this part. (In fact, I think that might be a good idea, now that you've pointed out that it can fail, to remove the audio source from the default config entirely.)

kpreid, 2016-05-07:
As of commit b8ca3fd8859d06b2e5ba18ada534f3d20a874c2e, `shinysdr --create` will now not put an `AudioDevice` in the generated config if creating one doesn't succeed.

closed 2016-05-07 by kpreid