kpreid/shinysdr #27: Lost the receiver!

k9wkj, 2015-03-10:
Great work so far!
I was playing around with a remote setup and I clicked the X in the receiver and it closed (well duhhh)
But now I cannot for the life of me get it back
Even went so far as to setup a separate folder to run from!
I've flushed browser history and tried my phone as well
Can't select a freq or change the center
I've purged and reinstalled twisted
Where did it go?

kpreid, 2015-03-11:
You get it back the same way you get it the first time: either click in the spectrum view on the left, or click on one of the frequency DB entries on the right. (You can shift-click to get more than one receiver at once.)

If clicking does nothing, then there's a bug of some sort. Please check the server output for a traceback or other messages that occur the moment you click.

k9wkj, 2015-03-11:
tried that again
and all i get is nuttin
the water fall flashes and the center frequency flashes
but nothing changes
here is the server talking from the launch to closing the browser tab

scannerman@scanner:~/Downloads/shinysdr$ python -m shinysdr.main configtest
INFO:shinysdr:Log opened.
linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown

gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
gr::log :INFO: audio source - Audio source arch: alsa
Using Volk machine: ssse3_64_orc
INFO:shinysdr:dbs/: Error opening database directory: OSError(2, 'No such file or directory')
INFO:shinysdr:Constructing flow graph...
INFO:shinysdr:Flow graph: Switching RF source
INFO:shinysdr:Flow graph: Changing sample rates
INFO:shinysdr:Flow graph: Rebuilding connections
INFO:shinysdr:Flow graph: ...done reconnecting.
INFO:shinysdr:Restoring state...
INFO:shinysdr:Flow graph: Switching RF source
INFO:shinysdr:Flow graph: Changing sample rates
INFO:shinysdr:Flow graph: Rebuilding connections
INFO:shinysdr:Flow graph: ...done reconnecting.
INFO:shinysdr:Starting web server...
INFO:shinysdr:OurStreamFactory (WebSocketFactory) starting on 8101
INFO:shinysdr:Starting factory <shinysdr.web.OurStreamFactory instance at 0x7fabefd216c8>
INFO:shinysdr:Starting factory <txws.WebSocketFactory instance at 0x7fabefd21710>
INFO:shinysdr:Site starting on 8100
INFO:shinysdr:Starting factory <twisted.web.server.Site instance at 0x7fabefcb11b8>
INFO:shinysdr:ShinySDR is ready.
INFO:shinysdr:Visit http://localhost:8100//
INFO:shinysdr:192.168.5.96 - - [11/Mar/2015:17:27:37 +0000] "GET //dbs/ HTTP/1.1" 200 39 "http://192.168.5.71:8100//" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/42.0.2311.11 Safari/537.36"
INFO:shinysdr:192.168.5.96 - - [11/Mar/2015:17:27:37 +0000] "GET //wdb/ HTTP/1.1" 200 2 "http://192.168.5.71:8100//" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/42.0.2311.11 Safari/537.36"
INFO:shinysdr:192.168.5.96 - - [11/Mar/2015:17:27:37 +0000] "GET /client/plugin-index.json HTTP/1.1" 200 215 "http://192.168.5.71:8100//" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/42.0.2311.11 Safari/537.36"
INFO:shinysdr:Starting RFC 6455 conversation
INFO:shinysdr:Stream connection to  //radio
INFO:shinysdr:Starting RFC 6455 conversation
INFO:shinysdr:Stream connection to  //audio?rate=44100
INFO:shinysdr:Flow graph: Rebuilding connections
INFO:shinysdr:Flow graph: ...done reconnecting.
INFO:shinysdr:Closing connection: '' (1001)
INFO:shinysdr:Closing connection: '' (1001)
INFO:shinysdr:Flow graph: Rebuilding connections
INFO:shinysdr:Flow graph: ...done reconnecting.

kpreid, 2015-03-11:
I see. Nothing looks wrong, but there should have been a POST log entry for creating the receiver. Please check the JavaScript console too (in Chrome, found under a "Developer" or "More tools" menu).

k9wkj, 2015-03-11:
i gited shinysdr to my desktop (gited ? gitted ? : the action of git cloning)
and it seems to run fine local on this machine with a dongle local
it does have a session closing bug that im going to try and get some good output on
ive experienced this on the broken server as well

now that is running on the wrong machine i have 5 recievers where ever i connect from
and it wont let my stop them
where is twisted storing its client preferences so i can axe the file?
so close
so very very close

thanks

 de k9wkj

kpreid, 2015-03-11:
The receiver settings are stored in the state file set in your configuration file. The default name is `state.json` in the current directory. (If you delete it make sure to also delete `state.json~` if it exists.)

(But if the X button doesn't delete the receiver, that's also a bug which needs looking at.)

k9wkj, 2015-03-11:
okay, i deleted the state.json and the temp file
and all the receivers are gone
and it wont create new ones
its in a very similar state to the other machine
page loads, spectrum runs and such  but no way to start a radio
im guessing the two ways to get to this point are closely related (closing a reciever/dumping the state file)

just tried creating a new config file  (--create x*blah*x )
and it didnt recreate the state file (thats why i had never noticed it before, it was already gone!)

k9wkj, 2015-03-11:
i found where/when things go wonkey
if you change the random goofy # on the url to "" (which i took to mean None)
it will not recreate the state file

now i can get to work and see which ways to get the bit rate slowed down
the dual core Atom im trying to use just hates this dsp work

k9wkj, 2015-03-11:
ohh does anyone have a example config file of how a HF upconverter should be used?
i really want to run 2 servers one for hf one for vhf/uhf
i may run out of cpu long before that.......

kpreid, 2015-03-11:
I'm not surprised the empty string doesn't work for the URL. I'll add a check for that (#28).

The Configuration section of the manual (default url http://localhost:8100/manual/configuration ) gives examples of how to configure for an upconverter. You don't need two servers; you can add the two devices to one server. (Unless you want to listen to both at once; that's not yet supported but will be soon.)

k9wkj, 2015-03-12:
im still not getting thru my skull how the config file comes together
i cant seem to get it to take a assigned key (rtl=0 or using the serial #)
nor will it take a different sample rate (2.4Ms is way to high for the slow atom)
ill post some input/output
also im having some issues with the server shutting down and even ending a session gracefully
if i close a tab it will drop into a unresponsive mode (this is from a remote machine) at which point i have to hunt down the pid and kill it by hand
the behaviour seems to have some bad effects, it even broke chrome badly enough i had to reinstall it
so a "start/stop" sending me data button im sure would fix this (thinking how gqrx operates)

but it ran all night with no issues!

kpreid, 2015-03-12:
> im still not getting thru my skull how the config file comes together  
> i cant seem to get it to take a assigned key (rtl=0 or using the serial #)
> nor will it take a different sample rate (2.4Ms is way to high for the slow atom)

Well, all I can say at this point is please show your configuration file and the server log when it fails. `OsmoSDRDevice('rtl=0')` ought to work regardless of anything else.

> also im having some issues with the server shutting down and even ending a session gracefully
> if i close a tab it will drop into a unresponsive mode (this is from a remote machine) at which point [...]

This definitely shouldn't happen. I haven't seen any reports of it before either. Frankly, either you're misdescribing the problem or it sounds like there's serious broad problems with your machine.

> so a "start/stop" sending me data button im sure would fix this (thinking how gqrx operates)

Nnnot really. The client drops the connections and the server acknowledges this and stops sending data — it's aware of the active connections, and even makes sure not to indefinitely buffer audio. If this doesn't work, again, something's badly broken in things I didn't write.

That said, I haven't tested on low-CPU configurations *recently* and it's possible there's some kind of unfortunate behavior there — I'll maybe dust off the BeagleBone and see what happens.

k9wkj, 2015-03-12:
ultimately i would like 2 dongles running on separate servers (yes i listen to many things at once from HF to UHF to Dr. Who audio books  all at once)
using 2 hf upconverters that gives me 2 hf and 2 vhf/uhf antenna inputs

 when i close a tab  i get

INFO:shinysdr:Closing connection: '' (1001)
INFO:shinysdr:Closing connection: '' (1001)
INFO:shinysdr:Flow graph: Rebuilding connections
INFO:shinysdr:Flow graph: ...done reconnecting.
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO^Z
[1]+  Stopped                 python -m shinysdr.main vhfdev4config
scannerman@scanner:~/Downloads/shinysdr$ 

i stopped it ctrl+z which leaves some dead process fodder that wont let you restart
so i have to hunt those down and kill them

i can close it ctrl+c local or ssh  but most times if i close a tab i get OOOOOOOOOOOOOOOO adinfinitum 

_______________________________________________________________________________________________

if i launch with this config

import shinysdr.devices
import shinysdr.plugins.osmosdr
import shinysdr.plugins.simulate

config.devices.add(u'osmo', shinysdr.plugins.osmosdr.OsmoSDRDevice('rtl=0'))

from shinysdr.plugins.osmosdr import OsmoSDRSource
config.devices.add(u'rtl', OsmoSDRSource('rtl=0', sample_rate=1200000, correction_ppm=7))

config.persist_to_file('state.json')

config.databases.add_directory('dbs/')

config.serve_web(
    http_endpoint='tcp:8100',
    ws_endpoint='tcp:8101',
    root_cap='K9WKJ',
    title='vhfdev4config')

i get

scannerman@scanner:~/Downloads/shinysdr$ python -m shinysdr.main vhfdev4config
INFO:shinysdr:Log opened.
linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown

gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000004
Found Rafael Micro R820T tuner
gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000004
usb_claim_interface error -6

FATAL: Failed to open rtlsdr device.

Trying to fill up 1 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

ERROR:shinysdr:main function encountered error
Traceback (most recent call last):
  File "/home/scannerman/Downloads/shinysdr/shinysdr/main.py", line 52, in main
    react(go)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/task.py", line 875, in react
    finished = main(_reactor, *argv)
  File "/home/scannerman/Downloads/shinysdr/shinysdr/main.py", line 47, in go
    return _main_async(reactor, argv, _abort_for_test)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/defer.py", line 1237, in unwindGenerator
    return _inlineCallbacks(None, gen, Deferred())
--- <exception caught here> ---
  File "/usr/lib/python2.7/dist-packages/twisted/internet/defer.py", line 1099, in _inlineCallbacks
    result = g.send(result)
  File "/home/scannerman/Downloads/shinysdr/shinysdr/main.py", line 91, in _main_async
    execute_config(configObj, args.configFile)
  File "shinysdr/config.py", line 211, in execute_config
    execfile(config_file, env)
  File "vhfdev4config", line 9, in <module>
    config.devices.add(u'rtl', OsmoSDRSource('rtl=0', sample_rate=1200000, correction_ppm=7))
  File "shinysdr/plugins/osmosdr.py", line 129, in OsmoSDRDevice
    raise LookupError('OsmoSDR device not found (device string = %r)' % osmo_device)
exceptions.LookupError: OsmoSDR device not found (device string = 'rtl=0')

INFO:shinysdr:Main loop terminated.
scannerman@scanner:~/Downloads/shinysdr$ 
____________________________________________________________________________________________________________

this is what i want to work or something similar  (i would like to use the serial # as a key for osmo to use)

import shinysdr.devices
import shinysdr.plugins.osmosdr
import shinysdr.plugins.simulate

config.devices.add(u'osmo', shinysdr.plugins.osmosdr.OsmoSDRDevice('rtl=0'))

from shinysdr.plugins.osmosdr import OsmoSDRSource
config.devices.add(u'rtl', OsmoSDRSource('rtl=0', sample_rate=250000, correction_ppm=7))
from shinysdr.devices import FrequencyShift
config.devices.add(u'rtl',
    OsmoSDRSource('rtl=0'),
    FrequencyShift(-55e6))

config.persist_to_file('state.json')

config.databases.add_directory('dbs/')

config.serve_web(
    http_endpoint='tcp:8100',
    ws_endpoint='tcp:8101',
    root_cap='K9WKJ',
    title='HFgoodness')

if i can get that one to run
im sure i can handle the rest

kpreid, 2015-03-12:
> when i close a tab i get
> 
> OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO

The OOOs are gr-osmosdr complaining that nobody's taking the samples (buffer overrun), which is because I stop all the DSP to save CPU when no clients are connected. That's unfortunately normal (I can't suppress them). But recent versions of gr-osmosdr (or is it librtlsdr, I'm not sure) don't do that, so you may have a rather old version of gr-osmosdr installed.

But, again, the OOOs are not a sign of a problem _if there is no client connected_. They should stop as soon as you reconnect.

> i stopped it ctrl+z which leaves some dead process fodder that wont let you restart

Well, yes, that's what ^Z does, suspend a process. There's no reason you should need to ^Z here. If it isn't responding to ^C (it should), try ^\ for a harder exit.

> if i launch with this config
>
>      config.devices.add(u'osmo', shinysdr.plugins.osmosdr.OsmoSDRDevice('rtl=0'))
>      config.devices.add(u'rtl', OsmoSDRSource('rtl=0', sample_rate=1200000, correction_ppm=7))

You're trying to open the same device (`rtl=0`) twice. That's why you get an error. If you're trying to use a second dongle, change one of the lines to `rtl=1` to specify the second device, or better yet use the devices' serial numbers instead of indexes. If you're trying to use only one dongle, then delete one of the lines.

k9wkj, 2015-03-13:
a bit of progress
this

import shinysdr.devices
import shinysdr.plugins.osmosdr
import shinysdr.plugins.simulate

from shinysdr.plugins.osmosdr import OsmoSDRSource
config.devices.add(u'rtl', OsmoSDRSource('rtl=0', sample_rate=250000, correction_ppm=7))
from shinysdr.devices import FrequencyShift
config.devices.add(FrequencyShift(-55e6))

config.persist_to_file('state.json')

config.serve_web(
    http_endpoint='tcp:8100',
    ws_endpoint='tcp:8101',
    root_cap='K9WKJ',
    title='hfdev4config')

gets me

scannerman@scanner:~/Downloads/shinysdr$ python -m shinysdr.main hfdev4config
INFO:shinysdr:Log opened.
linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown

gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000004
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
INFO:shinysdr:Constructing flow graph...
Using Volk machine: ssse3_64_orc
INFO:shinysdr:Flow graph: Switching RF source
INFO:shinysdr:Flow graph: Changing sample rates
INFO:shinysdr:Flow graph: Rebuilding connections
INFO:shinysdr:Flow graph: ...done reconnecting.
INFO:shinysdr:Restoring state...
INFO:shinysdr:Starting web server...
INFO:shinysdr:OurStreamFactory (WebSocketFactory) starting on 8101
INFO:shinysdr:Starting factory <shinysdr.web.OurStreamFactory instance at 0x7f3c3693db00>
INFO:shinysdr:Starting factory <txws.WebSocketFactory instance at 0x7f3c368ce9e0>
INFO:shinysdr:Site starting on 8100
INFO:shinysdr:Starting factory <twisted.web.server.Site instance at 0x7f3c29318cf8>
INFO:shinysdr:ShinySDR is ready.
INFO:shinysdr:Visit http://localhost:8100/K9WKJ/

but......
the offset isnt working
wether it is the dongle not wanting to go below its hard coded limit (which one can disable in Gqrx)
or it really isnt doing the (FrequencyShift(-55e6)
here is the lowest center freq i can get running that config
[image]

ohh my cpu use worries are not a trouble with the much lower sample bandwidth ,seems to hang about 40%

kpreid, 2015-03-14:
>     config.devices.add(u'rtl', OsmoSDRSource('rtl=0', sample_rate=250000, correction_ppm=7))
>     config.devices.add(FrequencyShift(-55e6))

The second line isn't doing anything useful at all (especially as there's no name specified). You need to add the frequency shift as *part* of the `'rtl'` device, so that the frequency shift is combined with the OsmoSDRSource and so modifies it:

     config.devices.add(u'rtl',
         OsmoSDRSource('rtl=0', sample_rate=250000, correction_ppm=7),
         FrequencyShift(-55e6))

Once you do this, the tuning range in the UI will automatically be adjusted to match.

k9wkj, 2015-03-14:
success !
 after a couple typos
this is working well

import shinysdr.devices
import shinysdr.plugins.osmosdr
import shinysdr.plugins.simulate

from shinysdr.plugins.osmosdr import OsmoSDRSource
from shinysdr.devices import FrequencyShift
config.devices.add(u'rtl',
     OsmoSDRSource('rtl=0', sample_rate=250000, correction_ppm=7),
     FrequencyShift(-55e6))

config.persist_to_file('state.json')

config.serve_web(
    http_endpoint='tcp:8100',
    ws_endpoint='tcp:8101',
    root_cap='K9WKJ',
    title='hfdev4config')

im sure i could trim some cruft from the file
but it runs fine
and displays the freq fine

Kevin i see by the manual that your thinking quite a ways ahead with this
BRAVO!!
ive been yelling for years that monolithic clumps of code are so 20th century
and that the browser can take care of most of our needs
and not just radio stuff (im working with the dev of the android app that communictes with a shooting chronograph  to get a browser app/extension/plugin to let us edit files off of the phone

sorry i was so thick skulled with this
but give a old fart a break.....

thanks for this!
if i can help out someway  yell!

now for adjustable filters........

k9wkj, 2015-03-14:
forgot
here is the happy evidence

[image]

kpreid, 2015-03-17:
Glad to hear it's working.

closed 2015-03-17 by kpreid