kpreid/shinysdr #30: exceptions.AttributeError: 'module' object has no attribute 'rotator_cc'

vo1pwf, 2015-04-06:
the pi2 doesnt have enough horsepower to do sdr well.... so I flattened a machine and put a fresh linux mint install on it and did the following:

sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
reboot
sudo apt-get install git git-core cmake libusb-1.0-0-dev build-essential libc6-dev  gnuradio  gr-osmosdr python-setuptools python-dev python-numpy
git clone https://github.com/kpreid/shinysdr.git
./fetch-js-deps.sh
python setup.py install
shinysdr --create shinysdr.config
shinysdr shinysdr.config 

but I am getting the following error:
INFO:shinysdr:Log opened.
linux; GNU C++ version 4.8.2; Boost_105400; UHD_003.005.005-0-unknown

gr-osmosdr 0.1.1 (0.1.1) gnuradio 3.7.2.1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
Home directory not accessible: Permission denied
Using Volk machine: ssse3_64_orc
ERROR:shinysdr:main function encountered error
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 54, in main
    react(go)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-15.0.0-py2.7-linux-x86_64.egg/twisted/internet/task.py", line 875, in react
    finished = main(_reactor, *argv)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 49, in go
    return _main_async(reactor, argv, _abort_for_test)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-15.0.0-py2.7-linux-x86_64.egg/twisted/internet/defer.py", line 1253, in unwindGenerator
    return _inlineCallbacks(None, gen, Deferred())
--- <exception caught here> ---
  File "/usr/local/lib/python2.7/dist-packages/Twisted-15.0.0-py2.7-linux-x86_64.egg/twisted/internet/defer.py", line 1107, in _inlineCallbacks
    result = g.send(result)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 93, in _main_async
    execute_config(configObj, args.configFile)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/config.py", line 226, in execute_config
    execfile(config_file, env)
  File "shinysdr.config", line 18, in <module>
    config.devices.add(u'sim', shinysdr.plugins.simulate.SimulatedDevice())
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/simulate.py", line 51, in SimulatedDevice
    rx_driver=_SimulatedRXDriver(name))
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/simulate.py", line 115, in __init__
    add_modulator(0.0, 'usb', 'USB')
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/simulate.py", line 103, in add_modulator
    tx = _SimulatedTransmitter(modulator, audio_rate, rf_rate, freq)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/simulate.py", line 201, in __init__
    self.__rotator = blocks.rotator_cc(rotator_inc(rate=rf_rate, shift=freq))
exceptions.AttributeError: 'module' object has no attribute 'rotator_cc'

INFO:shinysdr:Main loop terminated.
Reattached kernel driver

anyone have any ideas??

kpreid, 2015-04-07:
Your version of GNU Radio is just slightly too old. `gnuradio.blocks.rotator_cc` was added in version 3.7.2.2.

I see the README is wrong, so I'll update the minimum version specified there, but in any case I recommend using the latest release.

closed 2015-04-07 by kpreid

vo1pwf, 2015-04-07:
ok thanks...Ill give that a try