kpreid/shinysdr #31: crash on Loading client… Loading plugins… Connecting to server…

vo1pwf, 2015-04-11:
installed on fresh Linux Mint 17.1 Rebecca fine but however on trying to access it via chrome it crashes...steps I used to install is as follows and the errors below

sudo apt-get update
sudo apt-get upgrade

sudo apt-get install python-setuptools gr-osmosdr git git-core cmake libusb-1.0-0-dev build-essential libc6-dev libcppunit-1.13-0 libcppunit-dev libcppunit-subunit-dev libcppunit-subunit0 libgsl0-dev libgsl0ldbl python-cheetah g++ python-dev swig pkg-config libfftw3-dev libboost1.55-all-dev libcppunit-dev libgsl0-dev libusb-dev libsdl1.2-dev python-wxgtk2.8 python-numpy python-cheetah python-lxml doxygen libxi-dev python-sip libqt4-opengl-dev libqwt-dev libfontconfig1-dev libxrender-dev python-sip python-sip-dev

git clone git://git.osmocom.org/rtl-sdr.git
cd rtl-sdr; mkdir build; cd build
cmake ../ -DINSTALL_UDEV_RULES=ON -DDETACH_KERNEL_DRIVER=ON
make
sudo make install
sudo ldconfig
sudo  cp  ~/rtl-sdr/rtl-sdr.rules /etc/udev/rules.d/
rtl_test  -t 
(the test shows the RTL device...)

export PYTHONPATH=/usr/lib/python2.7/dist-packages

git clone --recursive http://git.gnuradio.org/git/gnuradio.git

cd gnuradio
mkdir build
cd build
cmake ../
make
make test
(tests all come back passed)
sudo make install

git clone https://github.com/kpreid/shinysdr.git
cd shinysdr
./fetch-js-deps.sh
sudo python setup.py install
python -m shinysdr.main --create ~/.shinysdr_settings
python -m shinysdr.main ~/.shinysdr_settings

INFO:shinysdr:Log opened.
linux; GNU C++ version 4.8.2; Boost_105400; UHD_003.005.005-0-unknown

gr-osmosdr 0.1.1 (0.1.1) gnuradio 3.7.2.1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
INFO: Audio source arch: alsa
Home directory not accessible: Permission denied
Using Volk machine: ssse3_64
Using Volk machine: ssse3_64_orc
INFO:shinysdr:dbs/: Error opening database directory: OSError(2, 'No such file or directory')
INFO:shinysdr:Constructing flow graph...
INFO:shinysdr:Flow graph: Switching RF source
INFO:shinysdr:Flow graph: Changing sample rates
INFO:shinysdr:Flow graph: Rebuilding connections
INFO:shinysdr:Flow graph: ...done reconnecting.
INFO:shinysdr:Restoring state...
INFO:shinysdr:Flow graph: Switching RF source
INFO:shinysdr:Flow graph: Changing sample rates
INFO:shinysdr:Flow graph: Rebuilding connections
INFO:shinysdr:Flow graph: ...done reconnecting.
INFO:shinysdr:Starting web server...
INFO:shinysdr:OurStreamFactory (WebSocketFactory) starting on 8101
INFO:shinysdr:Starting factory <shinysdr.web.OurStreamFactory instance at 0x7f705ff5ecf8>
INFO:shinysdr:Starting factory <txws.WebSocketFactory instance at 0x7f705ff5ed40>
INFO:shinysdr:Site starting on 8100
INFO:shinysdr:Starting factory <twisted.web.server.Site instance at 0x7f705ff0d7e8>
INFO:shinysdr:ShinySDR is ready.
INFO:shinysdr:Visit http://localhost:8100/None/
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOINFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /None/ HTTP/1.1" 200 3729 "-" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
OINFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/ui.css HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/theme.css HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/openlayers/OpenLayers.debug.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/require.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
OINFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /favicon.ico HTTP/1.1" 404 145 "-" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/main.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/values.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/events.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/database.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/network.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/maps.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/widget.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/widgets.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/audio.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/window-manager.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
OINFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /None/dbs/ HTTP/1.1" 200 39 "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /None/wdb/ HTTP/1.1" 200 2 "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/plugin-index.json HTTP/1.1" 200 215 "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/plugins/shinysdr.plugins.mode_s/mode_s.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/plugins/shinysdr.plugins.vor/vor.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/plugins/shinysdr.plugins.aprs/aprs.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:"10.1.1.10" - - [11/Apr/2015:17:14:52 +0000] "GET /client/plugins/shinysdr.plugins.hamlib/hamlib.js HTTP/1.1" 304 - "http://10.1.1.43:8100/None/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36"
INFO:shinysdr:Starting RFC 6455 conversation
INFO:shinysdr:Stream connection to  /None/radio

vo1pwf, 2015-04-11:
adding in the java console in the browser output:
Loading plugins…
:8100/client/main.js:22 Connecting to server…
:8100/client/network.js:220 WebSocket connection to 'ws://10.1.1.43:8101/None/audio?rate=44100' failed: Error during WebSocket handshake: net::ERR_CONNECTION_RESET
:8100/client/network.js:241 Lost WebSocket connection ws://10.1.1.43:8101/None/radio - reason given: 
:8100/client/network.js:220 WebSocket connection to 'ws://10.1.1.43:8101/None/radio' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
:8100/client/network.js:220 WebSocket connection to 'ws://10.1.1.43:8101/None/audio?rate=44100' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
:8100/client/network.js:220 WebSocket connection to 'ws://10.1.1.43:8101/None/radio' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
:8100/client/network.js:220 WebSocket connection to 'ws://10.1.1.43:8101/None/audio?rate=44100' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
:8100/client/network.js:220 WebSocket connection to 'ws://10.1.1.43:8101/None/radio' failed: Error in connection

kpreid, 2015-04-12:
Observations: The client sees its first connection reset and later ones refused. The server sees an initially successful connection (with the proper URL) and then later ones fail. Neither the client nor the server seem to be malfunctioning paticularly.

Do you have any firewall or antivirus software in use on either machine? I believe some firewalls break WebSocket connections due to not understanding them.

vo1pwf, 2015-04-12:
no firewall or anything....first thing I checked for....and selinux .....its one of those it works and it doesnt and theres no real reason poking its eye out to say why

really odd....after about 10 attempts I found the install that works flawlessly, everything compiles no complains, all tests pass..now this :-S

kpreid, 2015-04-12:
It does seem plausible that SELinux could be disallowing something ShinySDR needs. I'm not familiar with the details, but does it have some kind of policy-violation log?

vo1pwf, 2015-04-12:
selinux isnt installed....not a firewall or IP filtration issue that I can see....

I am currently following the same steps to install it on a PI2 just to see if that will work....
I did first get it up and running....I did soo much fooling around to get it to work that I wanted to make sure what was needed / not needed

so I kept my steps and tried to reinstall, every install afterwards has failed...if I can ever get it working again Im just going to image the OS and reuse it rather than compile again...any further developments Ill leave here

k9wkj, 2015-04-12:
just a thought
is there enough horse power to run the client on the server? for a test
that would show which end of the wire the trouble lies

kpreid, 2016-02-08:
Closing due to inactivity.

closed 2016-02-08 by kpreid