kpreid/shinysdr #32: script fails to install twisted on PI 2

vo1pwf, 2015-04-12:
fresh os install....got the rtl-sdr driver installed...and gnuradio...ran sudo python setup.py install and I get the following
.
.
.
Installed /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg
Processing dependencies for ShinySDR==0.0.0
Searching for txws
Reading http://pypi.python.org/simple/txws/
Best match: txWS 0.9.1
Downloading https://pypi.python.org/packages/source/t/txWS/txWS-0.9.1.tar.gz#md5=d113910af0521ea62db8a0f3d7c63abb
Processing txWS-0.9.1.tar.gz
Running txWS-0.9.1/setup.py -q bdist_egg --dist-dir /tmp/easy_install-CM4pl0/txWS-0.9.1/egg-dist-tmp-omkqix
Checking .pth file support in .
/usr/bin/python -E -c pass
Searching for vcversioner
Reading http://pypi.python.org/simple/vcversioner/
Best match: vcversioner 2.14.0.0
Downloading https://pypi.python.org/packages/source/v/vcversioner/vcversioner-2.14.0.0.tar.gz#md5=7848a365ced9941053bc25d9a9f8f4b4
Processing vcversioner-2.14.0.0.tar.gz
Running vcversioner-2.14.0.0/setup.py -q bdist_egg --dist-dir /tmp/easy_install-CM4pl0/txWS-0.9.1/temp/easy_install-x0Qh8C/vcversioner-2.14.0.0/egg-dist-tmp-_X8F9V
zip_safe flag not set; analyzing archive contents...

Installed /tmp/easy_install-CM4pl0/txWS-0.9.1/vcversioner-2.14.0.0-py2.7.egg
zip_safe flag not set; analyzing archive contents...
Adding txWS 0.9.1 to easy-install.pth file

Installed /usr/local/lib/python2.7/dist-packages/txWS-0.9.1-py2.7.egg
Searching for twisted
Reading http://pypi.python.org/simple/twisted/
Best match: Twisted 15.0.0
Downloading https://pypi.python.org/packages/source/T/Twisted/Twisted-15.0.0.tar.bz2#md5=f52bfe112a3aa33dd1d323c7c1e99f16
Processing Twisted-15.0.0.tar.bz2
Running Twisted-15.0.0/setup.py -q bdist_egg --dist-dir /tmp/easy_install-W3BoGN/Twisted-15.0.0/egg-dist-tmp-XZ1L1E
cc1: error: unrecognized command line option ‘-fstack-protector-strong’
cc1: error: unrecognized command line option ‘-fstack-protector-strong’
error: Setup script exited with error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

k9wkj, 2015-04-12:
ohh  i see your working with a Pi2
which distro are you using?
i really didnt have any issue at all getting the server running on raspian
but what ive found is the inherent crappy timing issues in the pi lead to really awful performance trouble
even though im only seeing around 200% cpu load it still has long (5 seconds) of unresponsiveness
very hard to work with
and then there is the NOISY ethernet controller (incredible noise levels)
i was thinking that a Bananna pi might be better even though it has a much less powerful cpu
the ethernet controller as well as the sata controller are not tied to the usb buss
which might keep things flowing more smoothly

kpreid, 2015-04-12:
Please don't have meandering conversations in the issue tracker.

The original problem is a problem with installing Twisted, not ShinySDR.

closed 2015-04-12 by kpreid