kpreid/shinysdr #33: shinysdr on odroid

henrycaga, 2015-04-23:
hi, im receiving this error. what is your advice? thank you.

# sudo python -m shinysdr.main test.config
INFO:shinysdr:Log opened.
linux; GNU C++ version 4.8.2; Boost_105400; UHD_003.008.002-151-gbf4f50de

gr-osmosdr v0.1.4-16-g61184a19 (0.1.5git) gnuradio 3.7.6.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace
Using device #0 Generic RTL2832U SN: 77771111153705700
Found Rafael Micro R820T tuner
INFO: Audio source arch: alsa
Home directory not accessible: Permission denied
ERROR: [default]: No such file or directory
ERROR:shinysdr:main function encountered error
Traceback (most recent call last):
  File "/opt/shinysdr/shinysdr/main.py", line 54, in main
    react(go)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/task.py", line 875, in react
    finished = main(_reactor, *argv)
  File "/opt/shinysdr/shinysdr/main.py", line 49, in go
    return _main_async(reactor, argv, _abort_for_test)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/defer.py", line 1253, in unwindGenerator
    return _inlineCallbacks(None, gen, Deferred())
--- <exception caught here> ---
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/defer.py", line 1107, in _inlineCallbacks
    result = g.send(result)
  File "/opt/shinysdr/shinysdr/main.py", line 93, in _main_async
    execute_config(configObj, args.configFile)
  File "shinysdr/config.py", line 226, in execute_config
    execfile(config_file, env)
  File "test.config", line 15, in <module>
    config.devices.add(u'audio', shinysdr.devices.AudioDevice(''))
  File "shinysdr/devices.py", line 343, in AudioDevice
    quadrature_as_stereo=quadrature_as_stereo)
  File "shinysdr/devices.py", line 399, in __init__
    ok_to_block=True)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/audio/audio_swig.py", line 133, in make
    return _audio_swig.source_make(*args, **kwargs)
exceptions.RuntimeError: audio_alsa_source

INFO:shinysdr:Main loop terminated.

kpreid, 2015-04-23:
This issue is a duplicate of #26.

My advice is to remove the line

    config.devices.add(u'audio', shinysdr.devices.AudioDevice(''))

from the config. It's not needed unless you're using a soundcard SDR, which I assume you aren't.

closed 2015-04-23 by kpreid

kpreid, 2016-05-07:
As of commit b8ca3fd8859d06b2e5ba18ada534f3d20a874c2e, `shinysdr --create` will now not put an `AudioDevice` in the generated config if creating one doesn't succeed.