kpreid/shinysdr #34: USRP2 problem

dogthunder, 2015-06-19:
I just installed shinysdr ant server runs fine until I try to connect to it :+1: 
gr-osmosdr v0.1.4-29-g44c223cb (0.1.5git) gnuradio v3.7.7.1-120-g67463e74
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace 
-- Opening a USRP2/N-Series device...
-- Current recv frame size: 1472 bytes
-- Current send frame size: 1472 bytes
-- Using subdev spec 'A:AB'.
Using Volk machine: avx_64_mmx
INFO:shinysdr:dbs/: Error opening database directory: OSError(2, 'No such file or directory')
INFO:shinysdr:Constructing flow graph...
INFO:shinysdr:Flow graph: Switching RF device to osmo
INFO:shinysdr:Flow graph: Rebuilding connections because: initialization, switched device
INFO:shinysdr:Flow graph: ...done reconnecting (4 ms).
INFO:shinysdr:Restoring state...
INFO:shinysdr:Starting web server...
INFO:shinysdr:OurStreamFactory (WebSocketFactory) starting on 8101
INFO:shinysdr:Starting factory <shinysdr.web.OurStreamFactory instance at 0x7f8c5835e8c0>
INFO:shinysdr:Starting factory <txws.WebSocketFactory instance at 0x7f8c5835e950>
INFO:shinysdr:Site starting on 8100
INFO:shinysdr:Starting factory <twisted.web.server.Site instance at 0x7f8c58369758>
INFO:shinysdr:ShinySDR is ready.
INFO:shinysdr:Visit http://localhost:8100/T8BO2WdC56UBqFgKeQA1BA/
INFO:shinysdr:Starting RFC 6455 conversation
INFO:shinysdr:Stream connection to  /T8BO2WdC56UBqFgKeQA1BA/audio?rate=44100
INFO:shinysdr:Flow graph: Rebuilding connections because: added audio queue
INFO:shinysdr:Flow graph: ...done reconnecting (0 ms).
INFO:shinysdr:Starting RFC 6455 conversation
INFO:shinysdr:Stream connection to  /T8BO2WdC56UBqFgKeQA1BA/radio
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 4761904. Due to alignment requirements
   256 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::buffer::allocate_buffer: failed to allocate buffer of size 1190476 KB
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 4761904. Due to alignment requirements
   256 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::buffer::allocate_buffer: failed to allocate buffer of size 1190476 KB
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

I increased my shmmax 
cat /proc/sys/kernel/shmmax
4294967296

but still the same.  Osmocom source works fine from GRC.

kpreid, 2015-06-19:
Note that 1190276 KiB is 1.1 GiB. So it's probably failing due to multiple allocations of that size, and even if the shared memory limit were big enough, you wouldn't want to actually operate under that condition. I'm going to guess that the OsmoSDR driver is defaulting to a very large sample rate and at least one of ShinySDR's buffers is scaling with the sample rate (`shinysdr.top.MaxProbe` is one I know does).

If you haven't already, I recommend setting a very low sample rate in your config file — say, 2 MS/s, or whatever is the lower limit for your USRP — and see if that works, then increase it.

    config.devices.add(u'usrp', OsmoSDRSource('uhd', sample_rate=2000000))

Let me know what happens, and if the sample rate is clearly the problem, then I should start trying to diagnose what parts of ShinySDR are using too large buffers.

kpreid, 2015-09-13:
Changed MaxProbe to use a power-of-2 size in commit d3a6c7b987fb05f66509827283d3a6a4aabfb33d.

kpreid, 2016-02-08:
Nothing heard recently, and I expect the MaxProbe change fixed/improved this.

closed 2016-02-08 by kpreid