kpreid/shinysdr #35: the problem about installation

mitchell520, 2015-07-09:
hi,when i install you shinysdr,I encounter problems:
mitchell@mitchell-desktop:~/sdr/shinysdr$ sudo shinysdr 1
INFO:shinysdr:Log opened.
linux; GNU C++ version 4.8.2; Boost_105400; UHD_003.005.005-0-unknown

gr-osmosdr 0.1.1 (0.1.1) gnuradio 3.7.2.1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
Using Volk machine: sse4_2_64_orc
ERROR:shinysdr:main function encountered error
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 54, in main
    react(go)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/task.py", line 867, in react
    finished = main(_reactor, *argv)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 49, in go
    return _main_async(reactor, argv, _abort_for_test)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/defer.py", line 1237, in unwindGenerator
    return _inlineCallbacks(None, gen, Deferred())
--- <exception caught here> ---
  File "/usr/lib/python2.7/dist-packages/twisted/internet/defer.py", line 1099, in _inlineCallbacks
    result = g.send(result)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 93, in _main_async
    execute_config(configObj, args.configFile)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/config.py", line 226, in execute_config
    execfile(config_file, env)
  File "1", line 18, in <module>
    config.devices.add(u'sim', shinysdr.plugins.simulate.SimulatedDevice())
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/simulate.py", line 52, in SimulatedDevice
    rx_driver=_SimulatedRXDriver(name))
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/simulate.py", line 117, in __init__
    add_modulator(0.0, 'usb', 'USB')
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/simulate.py", line 105, in add_modulator
    tx = _SimulatedTransmitter(modulator, audio_rate, rf_rate, freq)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/simulate.py", line 203, in __init__
    self.__rotator = blocks.rotator_cc(rotator_inc(rate=rf_rate, shift=freq))
exceptions.AttributeError: 'module' object has no attribute 'rotator_cc'

INFO:shinysdr:Main loop terminated.
 Can you tell me how can I solve it?

k9wkj, 2015-07-09:
> unwindGenerator return _inlineCallbacks(None, gen, Deferred()) --- --- File
> in config.devices.add(u'sim', shinysdr.plugins.simulate.SimulatedDevice())

does it run if you dont install it?
like
mitchell@mitchell-desktop:~/sdr/shinysdr python -m shinysdr.main 1

i had a similar problem to yours
so rather than chase it i just run it right from the shinysdr dir

i bet Kevin has seen this and can really help you

kpreid, 2015-07-09:
Your version of GNU Radio is just slightly too old. gnuradio.blocks.rotator_cc was added in version 3.7.2.2. I recommend using the latest release of GNU Radio.

(duplicate of #30)

closed 2015-07-09 by kpreid