kpreid/shinysdr #37: Implement DSD "demodulator" with gr-dsd

ryanturner, 2015-08-03:
There's a great project https://github.com/argilo/gr-dsd that works reasonably well. I'm looking through the source and I may take a stab at integrating this myself. Here's an example of something using gr-dsd: https://github.com/robotastic/gr-dsd/blob/master/p25_dsd_rx.py

Anyway, this would be a big improvement in my opinion; the one additional demodulator would allow for dmr, nxdn, p25p1, and x2tdma voice demodulation.

kpreid, 2015-08-03:
That was fairly easy. Done in commit ec832315a5f4dc8cacea72ea595b943a811576f1.

Unfortunately, `dsd.block_ff` has a severe lack of methods: it neither allows changing the parameters live (only at block construction time) nor reading back any status information about whether it is successfully demodulating (such as dsd internally prints to stdout), so you have to use it sort of blind (other than hearind the audio). If you're interested in this feature, please consider suggesting/contributing such improvements to upstream dsd / gr-dsd.

closed 2015-08-03 by kpreid

ryanturner, 2015-08-03:
Wow, awesome work. This was not quite the path I started down for implementation, so I'm glad you came in to help.

kpreid, 2015-08-03:
I had to use a different version, too. That's why I didn't link anything in the README — I'll wait for upstream to get its act together before saying who's upstream.

It would be informative to hear what “path you started down”. (Adding an option to the basic FM demodulator?) ShinySDR is not exactly ideally architected and I'd like to know what looked like a good idea because there's always a chance it might be a better idea in the long run.