kpreid/shinysdr #38: Raspberry Pi 2 - rtlsdr_read_async returned with -5

corpr8, 2015-08-14:
Hi there, 

I've installed Shinysdr on a raspberry pi 2 running Raspbian Wheezy - (and it looks really promising). I got rid of Alsa from the config file (as per our previous thread) - however, when I try and retune from the web interface, I get the following messages and it hangs the session:

rtlsdr_read_async returned with -5
INFO:shinysdr:Flow graph: ...done reconnecting (20 ms).
INFO:shinysdr:192.168.1.65 - - [14/Aug/2015:20:34:25 +0000] "PUT /51sSBRbBTwHsM6xR-4WJ2g/radio/sources/osmo/rx_driver/agc HTTP/1.1" 204 - "http://ham:8100/51sSBRbBTwHsM6xR-4WJ2g/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/44.0.2403.155 Safari/537.36"
INFO:shinysdr:Constructed AM demodulator: 124 ms.
rtlsdr_read_async returned with -5

Any idea's on how to fix this? 

Many thanks!

k9wkj, 2015-08-14:
the bad news is
this will not  run well on a Pi2
ive tried everything i can think of  and there just isnt enough Oommph in a
Pi
i have it running on a 1.6ghz dual core Atom and thats about as light as it
will run on in a useable fashion
in my experiments anyway
maybe Kevin will have some more insight

corpr8, 2015-08-14:
Ah ok - thanks for the heads up!

corpr8, 2015-08-14:
Fixed - with the link: http://photobyte.org/rtl-sdr-segmentation-error-solved/

Essentially libusb needed updating via sudo apt-get install libusb-1.0

closed 2015-08-14 by corpr8