kpreid/shinysdr #39: Add screenshots in readme

qwitwa, 2015-09-24:
Since the selling-point of this project seems to be ease of use/having a nice UI, it seems weird not to demonstrate this non-clunkiness in the readme with a couple of screenshots. I'd be far more likely to install and mess about with this if I knew what it looked like in advance.

andyinabox, 2015-10-16:
+1 on this!

kpreid, 2015-10-16:
Just because it's easy to use doesn't mean it looks good! But seriously, I agree that this would be a good thing.

But because it would be Wrong™ in my opinion to put such things as screenshots in the repository (since doing so permanently increases the total size and it's more like a build product than a source file), this'll be on hold until I get around to making the radio part of my main web site, or more generally any sort of “marketing page”, if you will, for ShinySDR.

kpreid, 2016-02-08:
My current thinking is: There should be a web site _about_ ShinySDR, located at `shinysdr.switchb.org`. It should contain at least:

* A description of ShinySDR, similar to the current README, with the addition of screenshots and any other useful “marketing” information.
* A copy of the manual.
* Installation instructions, possibly integrated into the manual (since the manual would now be readily available to people who haven't installed yet).
* `/tools/audio-fft`, as a showcase piece.

This suggests that the way this web site should be _built_ is by deriving it from the `shinysdr/webstatic/` tree, so that the URLs to the manual and such are identically structured, and adding an `index.html` with the “marketing” content.

kpreid, 2016-06-18:
Have written the basic site content https://github.com/kpreid/shinysdr-site but still need to:

* Host it.
* Add some screenshots. (I keep not wanting to because I see all the not-really-finished things in the UI.)
* Move installation instructions into the site or manual, and remove now-duplicate content from the README in this repository.
* Link to `/tools/audio-fft` (probably create a tools page in webstatic so as not to keep several link-list pages in sync).

k9wkj, 2016-06-18:
hey Kevin
ive got some screen shots you can use
showing wide screen use
and a shot using 6 or so P25 demodulators via DSD  (youll see why the
receiver tabs on the bottom would be awesome!!)

http://i.imgur.com/N3lsb6N.jpg
http://i.imgur.com/ACcWPhO.png

if you need something special let me know

anarcat, 2017-04-22:
> Host it.

I encourage you to look at github pages - it's simple and just works and removes another burden from your shoulders...

kpreid, 2017-04-22:
...I was going to object that I have a perfectly good web server, but then I realized that it isn't perfect in that it isn't configured for HTTPS (yet) (which is relevant for some of the 'demo' stuff I want to do) so GitHub Pages is a fine interim location. Thank you for the idea.

anarcat, 2017-04-22:
it's all fine until you get on slashdot or hackernews or something. ;) better be someone else's problem here...

kpreid, 2017-04-22:
- [x] **https://kpreid.github.io/shinysdr/ now exists.**
- [ ] Add some screenshots — or perhaps better, screencast (video)!
- [ ] Move installation instructions into the site or manual, and remove now-duplicate content from the README in this repository.
- [ ] Link to `/tools/` (probably create a tools page in webstatic so as not to keep several link-list pages in sync) as "powered by ShinySDR!" demos.

k9wkj, 2017-04-22:
looks nifty kevin !

kpreid, 2017-04-22:
I have now moved the installation instructions into the manual, and added more information: 746b85a66b6e3b63ef074f0a28f066d6246888c6

kpreid, 2017-04-23:
- [x] **Link to `/tools/` (probably create a tools page in webstatic so as not to keep several link-list pages in sync) as "powered by ShinySDR!" demos.**
- [x] **Move installation instructions into the site or manual, and remove now-duplicate content from the README in this repository.**
- [ ] Add some screenshots — or perhaps better, screencast (video)!
- [ ] **Come up with “branding” for ShinySDR — that is, things like colors, an icon, and a stylesheet for non-app pages.** (Filed #72,)