kpreid/shinysdr #41: Power User Features

k9wkj, 2015-11-17:
Since the HF band conditions have been a bit more stable the past weeks, and i got my power supply for the remote equipment a bit more nailed down
ive run across a thing that might be of benefit to a "power user"
here is a page grab of Shiny running 4 receivers in the 15m band, i had it stretched across both monitors to give it a bit more of a dramatic effect.
http://imgur.com/a/tdPe2
but as you can see one can only garner info from 2 to 3 of the recivers at once.
maybe a compact view with drop down mode selection menu instead of buttons, and maybe even just numbers for the agc,signal level and such.
or even a similar compact receiver across the bottom of the page.
so really just a simple view that allows more concentrated information.

please note the Ionosonde thats passing up band on this shot, pretty cool huh?
im such a radio geek.....

kpreid, 2015-11-18:
Thanks for the suggestions!

Hypothetically: How would you like, instead of having more full size controls, if you could click on the receiver marks on the spectrum to select one receiver to change?

k9wkj, 2015-11-18:
now thats a great idea!
click in the bandpass to bring up the controls!
would it be possible to have the mode and frequency info on the "flag" on
the receiver spectrum be in a bigger font
im pretty sure that most users wont want this control feature
as its a bit more confusing for folks that are not radio geeks!

i have a few other things i would like to talk about as well
really questions on things that may already be implemented  but im not sure
how they are used yet
and of course my constant begging for adjustable filters.....

this thing is really working well!
i still have the ooooooooo  thing  at times
but not often