kpreid/shinysdr #42: WebServer does not load

Kayzh3r, 2016-02-04:
Hi!!!
First of all, congratulations for your job.
The issue is simply; when i start the server it is loading client all time.

ignazio@apolo:~/Desktop/shinySDR$ shinysdr intento2
INFO:shinysdr:Log opened.
linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.git-119-g42a3eeb6

gr-osmosdr v0.1.4-67-gac15e789 (0.1.5git) gnuradio 3.7.9
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[R82XX] PLL not locked!
INFO: Audio source arch: alsa
Using Volk machine: avx_64_mmx_orc
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/aprs/dropin.cache: error number 13
gr-air-modes warning: numpy+scipy not installed, FlightGear interface not supported
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/mode_s/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/vor/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/hamlib/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/rtty/dropin.cache: error number 13
INFO:shinysdr:dbs/: Error opening database directory: OSError(2, 'No such file or directory')
INFO:shinysdr:Constructing...
INFO:shinysdr:Flow graph: Switching RF device to audio
INFO:shinysdr:Flow graph: Rebuilding connections because: initialization, switched device
INFO:shinysdr:Flow graph: ...done reconnecting (7 ms).
INFO:shinysdr:Restoring state...
INFO:shinysdr:Warning: Discarding nonexistent state <shinysdr.session.AppRoot object at 0x7f1fd9c44d50>.source_name = osmo 
INFO:shinysdr:Starting web server...
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/aprs/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/mode_s/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/vor/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/hamlib/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/rtty/dropin.cache: error number 13
INFO:shinysdr:OurStreamFactory (WebSocketFactory) starting on 8101
INFO:shinysdr:Starting factory <shinysdr.web.OurStreamFactory instance at 0x7f1fc6287c68>
INFO:shinysdr:Starting factory <txws.WebSocketFactory instance at 0x7f1fc6287cb0>
INFO:shinysdr:Site starting on 8100
INFO:shinysdr:Starting factory <twisted.web.server.Site instance at 0x7f1fc624f7a0>
INFO:shinysdr:ShinySDR is ready.
INFO:shinysdr:Visit http://localhost:8100/kbvjf53fCd9EV9CrJOIG_Q/
INFO:shinysdr:"127.0.0.1" - - [04/Feb/2016:20:26:01 +0000] "GET /kbvjf53fCd9EV9CrJOIG_Q/ HTTP/1.1" 200 4090 "-" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Ubuntu Chromium/48.0.2564.82 Chrome/48.0.2564.82 Safari/537.36"
INFO:shinysdr:"127.0.0.1" - - [04/Feb/2016:20:26:01 +0000] "GET /client/require.js HTTP/1.1" 404 145 "http://localhost:8100/kbvjf53fCd9EV9CrJOIG_Q/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Ubuntu Chromium/48.0.2564.82 Chrome/48.0.2564.82 Safari/537.36"
INFO:shinysdr:"127.0.0.1" - - [04/Feb/2016:20:26:08 +0000] "GET /kbvjf53fCd9EV9CrJOIG_Q/ HTTP/1.1" 200 4090 "-" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Ubuntu Chromium/48.0.2564.82 Chrome/48.0.2564.82 Safari/537.36"
INFO:shinysdr:"127.0.0.1" - - [04/Feb/2016:20:26:08 +0000] "GET /client/require.js HTTP/1.1" 404 145 "http://localhost:8100/kbvjf53fCd9EV9CrJOIG_Q/" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Ubuntu Chromium/48.0.2564.82 Chrome/48.0.2564.82 Safari/537.36"

kpreid, 2016-02-05:
See that last HTTP 404 log entry? You didn't run fetch-js-deps.sh before installing.

closed 2016-02-05 by kpreid

Kayzh3r, 2016-02-05:
Thanks so much...sorry...it was very foolish

kpreid, 2016-02-05:
You're not the only one to make this mistake. I'm now planning to expand the dependency checks at startup so that there's a nice error message for this case.