kpreid/shinysdr #46: IQ Reversed on HackRF

aldude999, 2016-04-24:
I've noticed on the HackRF, there doesn't seem to be a way to filp the IQ, and it seems to be reversed. Does anybody know how I could flip the IQ with ShinySDR?

devnulling, 2016-04-24:
@aldude999 Have you updated the CPLD & firmware on the HackRF? There was a commit ~2 years ago that flipped the IQ in the HackRF CPLD. 

https://github.com/mossmann/hackrf/wiki/Updating-Firmware

aldude999, 2016-04-24:
Yes, I am on the newest version, hackrf_info reports
`Found HackRF board 0:
USB descriptor string: 0000000000000000457863c825272d1f
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2
Part ID Number: 0xa000cb3c 0x00584353
Serial Number: 0x00000000 0x00000000 0x457863c8 0x25272d1f
`

devnulling, 2016-04-25:
@aldude999 Did you update the CPLD image as well, and not just the firmware? (Step 2 in the link above). I ran into this same issue with IQ being flipped, because the CPLD image wasn't updated.

devnulling, 2016-04-25:
Just for reference, the relevant git commit - https://github.com/mossmann/hackrf/commit/8e387e5489077d838e889179612c266830ed3913

aldude999, 2016-04-25:
Thanks so much for the help, I guess I never flashed the CPLD file when I updated the firmware. Flashed it and now it's working with no problems!

closed 2016-04-25 by aldude999

kpreid, 2016-04-25:
@devnulling Thanks for helping!