kpreid/shinysdr #49: remote execution  of shinysdr

SoufianLa, 2016-05-07:
INFO:shinysdr:Log opened.
linux; GNU C++ version 4.9.2; Boost_105500; UHD_003.007.003-0-unknown

gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[R82XX] PLL not locked!
gr::log :INFO: audio source - Audio source arch: alsa
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Access denied

gr::log :ERROR: audio_alsa_source0 - [default]: Connection refused
ERROR:shinysdr:main function encountered error
Traceback (most recent call last):
  File "/root/shinysdr/shinysdr/main.py", line 55, in main
    react(go)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/task.py", line 875, in react
    finished = main(_reactor, *argv)
  File "/root/shinysdr/shinysdr/main.py", line 50, in go
    return _main_async(reactor, argv, _abort_for_test)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/defer.py", line 1237, in unwindGenerator
    return _inlineCallbacks(None, gen, Deferred())
--- <exception caught here> ---
  File "/usr/lib/python2.7/dist-packages/twisted/internet/defer.py", line 1099, in _inlineCallbacks
    result = g.send(result)
  File "/root/shinysdr/shinysdr/main.py", line 101, in _main_async
    execute_config(configObj, args.configFile)
  File "shinysdr/config.py", line 234, in execute_config
    execfile(config_file, env)
  File "configuration", line 15, in <module>
    config.devices.add(u'audio', AudioDevice(''))
  File "shinysdr/devices.py", line 374, in AudioDevice
    quadrature_as_stereo=quadrature_as_stereo)
  File "shinysdr/devices.py", line 430, in __init__
    ok_to_block=True)
  File "/usr/lib/python2.7/dist-packages/gnuradio/audio/audio_swig.py", line 133, in make
    return _audio_swig.source_make(*args, **kwargs)
exceptions.RuntimeError: audio_alsa_source

INFO:shinysdr:Main loop terminated.

kpreid, 2016-05-07:
If you do not have a functioning audio input device, remove the line 

    config.devices.add(u'audio', AudioDevice(''))

from your config file.

closed 2016-05-07 by kpreid

kpreid, 2016-05-07:
As of commit b8ca3fd8859d06b2e5ba18ada534f3d20a874c2e, `shinysdr --create` will now not put an `AudioDevice` in the generated config if creating one doesn't succeed.