kpreid/shinysdr #50: gnuradio.blocks not present?

ebenhaber, 2016-06-11:
I downloaded ShinySDR to my mac, installed the latest version of gnuradio, but I’m getting an odd error:

```
eben$ shinysdr --create ~/shiny.config
INFO:shinysdr:Log opened.
The following libraries/programs are missing:
	GNU Radio  (Check: gnuradio.blocks not present.)
The following libraries/programs are too old:
	GNU Radio  (Check: gnuradio.blocks.rotator_cc not present.)
Please (re)install current versions.
INFO:shinysdr:Main loop terminated.
```

But gnuradio is there:

```
eben$ sudo port list gnu radio
Password:
gnuradio                       @3.7.9.2        science/gnuradio
```

All my web searches for gnuradio.blocks suggest that it's part of gnuradio, and I don't see anything related to 'gnuradio.blocks' in the README file. 

Do you have any suggestions? 

Thanks!

kpreid, 2016-06-11:
> ```
> eben$ sudo port list gnu radio
> Password:
> gnuradio                       @3.7.9.2        science/gnuradio
> ```

This doesn't show that you have installed `gnuradio`, only that MacPorts knows that it is available. Use `port installed gnuradio` instead to check.

If you install `gnuradio` and ShinySDR still doesn't find it, try specifying the MacPorts version of Python when you install ShinySDR: `/opt/local/bin/python2.7 setup.py install`. This should set up the `shinysdr` command to use that version and it will know about the GNU Radio modules where Apple's provided `python` might not.

ebenhaber, 2016-06-13:
Thanks for the suggestions. It looks like it was a MacPorts issue - it claimed several versions of gnu radio were installed, so I uninstalled and reinstalled gnu radio and the packages that depend on it, and everything started working.

closed 2016-06-13 by ebenhaber