kpreid/shinysdr #52: ImportError: No module named dsd_swig

mattia-beta, 2016-08-28:
Hi, i've just built gr-dsd from source but the DSD demodulator is not showing inside shinysdr.
Looking at the source code I've found this snippet:

> try:
>     from dsd import block_ff as dsd_block_ff
>     _available = True
> except ImportError:
>     _available = False
 
I think the problem is that shinysdr can't find the dsd python module.
I've manually tried to import this module inside a python script but I got this error:

> from dsd import block_ff as dsd_block_ff
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "**/usr/lib/python2.7/dist-packages/**dsd/__init__.py", line 45, in <module>
>     from dsd_swig import *
> ImportError: No module named dsd_swig

I've also checked the PYTHON_PATH running:
`python -c "import sys; print '\n'.join(sys.path)"`

and the response is:

> /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg
> /usr/local/lib/python2.7/dist-packages/ephem-3.7.6.0-py2.7-linux-armv7l.egg
> /usr/local/lib/python2.7/dist-packages/txWS-0.9.1-py2.7.egg
> /usr/local/lib/python2.7/dist-packages/Twisted-16.3.2-py2.7-linux-armv7l.egg
> /usr/local/lib/python2.7/dist-packages/zope.interface-4.2.0-py2.7-linux-armv7l.egg
> **/usr/lib/python2.7/dist-packages**
> /usr/lib/python2.7
> /usr/lib/python2.7/plat-arm-linux-gnueabihf
> /usr/lib/python2.7/lib-tk
> /usr/lib/python2.7/lib-old
> /usr/lib/python2.7/lib-dynload
> /usr/local/lib/python2.7/dist-packages
> /usr/lib/python2.7/dist-packages/PILcompat
> /usr/lib/python2.7/dist-packages/gtk-2.0
> /usr/lib/pymodules/python2.7
> /usr/lib/python2.7/dist-packages/wx-3.0-gtk2

Searching for "gr-dsd" in the filesystem:

> /usr/lib/libgr-dsd.so
> /usr/local/lib/libgr-dsd.so

Any idea?
How can I fix this error?
Thank you =)

kpreid, 2016-08-28:
I'm afraid I don't have any advice for Python SWIG issues — and as you've already figured out, this is a problem with installing gr-dsd and not ShinySDR per se. Sorry!

closed 2016-08-28 by kpreid

mattia-beta, 2016-09-01:
@kpreid I've fixed this issue doing:

` cmake . -DPYTHON_INCLUDE_DIR=/usr/include/python2.7 -DPYTHON_LIBRARY=/usr/lib/libpyglib-2.0-python2.7.so.0 `

` make`

` sudo make install `

` sudo ldconfig`

Otherwise _dsd_swig.so is built and linked against libpython3.5m.so, leading to the following error on "import dsd": dynamic module does not define init function (init_dsd_swig)

Bye