kpreid/shinysdr #54: Audio under-run with USRP N200

mehdideveloper, 2016-10-27:
Hi
I'm using the latest shinysdr on OS X 10.12.1
It runs fine with RTL-SDR, but not with USRP N200. I have audio underrun and glitch. Also, there's no spectrum waterfall.
Here's a comparison between running RTL-SDR and USRP:

Any ideas?

My system info:
MacBook Pro 2015 on OS X 10.12.1
Python 2.7.11
shinysdr and gr-osmosdr: from git, latest.
GNURadio 3.7.9.1
Chrom version 52
USRP's device string in config: 
`config.devices.add(u'osmo', OsmoSDRDevice('uhd', sample_rate=250000))`

kpreid, 2016-10-27:
The missing waterfall probably means there's something else wrong, but does this sound like #51? In particular, is the audio underrunning in AM mode but OK in narrow FM?

Also, whatever you meant to show in “Here's a comparison…” didn't make it in.

mehdideveloper, 2016-10-27:
I updated the original post with two links (https://i.imgsafe.org/24f30d9aa4.png
https://i.imgsafe.org/24f2f9d786.png) to show you my waterfall.
It's similar to the issue #51 (almost always getting underron on FM and AM, but not on SSBs. On NFM, I get underrun once in a while)

kpreid, 2016-10-27:
OK, so that is probably the same as #51. It's a weird bug, and I haven't gotten around to digging into it yet.

Your waterfall display is working normally — it is just out of range, because the USRP's signal is at a different level than the RTL-SDR's. I can see some small peaks and faint blue in the screenshot. Click on the “Options” at the top right and adjust “Lowest value” until you can see the spectrum.

closed 2016-10-27 by kpreid