kpreid/shinysdr #59: No radio options

mehdideveloper, 2017-02-19:
Hi
I run shinysdr on my OSX without problems: I can see the spectrum on my Chrome, but no audio and there's no section to change receiver options like Modulation.
What can cause this?

Thanks

Output log:

> INFO:shinysdr:Log opened.
> Mac OS; Clang version 8.0.0 (clang-800.0.42.1); Boost_106300; UHD_003.010.000.000-0-unknown
> 
> gr-osmosdr v0.1.4-75-gae686c46 (0.1.5git) gnuradio 3.7.10.1
> built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya
> Using device #0 Realtek RTL2838UHIDIR SN: 00000001
> Found Rafael Micro R820T tuner
> [R82XX] PLL not locked!
> [R82XX] PLL not locked!
> INFO: Audio source arch: osx
> INFO:
> 
> Using input audio device 'Built-in Microphone'.
>   ... which is the current default input audio device.
>   Changing the default input audio device in the System Preferences will
>   result in changing it here, too (with an internal reconfiguration).
> 
> INFO:shinysdr:Constructing...
> INFO:shinysdr:Flow graph: Switching RF device to audio
> INFO:shinysdr:Flow graph: Rebuilding connections because: initialization, switched device
> INFO:shinysdr:Flow graph: ...done reconnecting (6 ms).
> INFO:shinysdr:Restoring state...
> INFO:shinysdr:Flow graph: Switching RF device to osmo
> INFO:shinysdr:Flow graph: Rebuilding connections because: switched device
> INFO:shinysdr:Flow graph: ...done reconnecting (5 ms).
> INFO:shinysdr:Starting web server...
> INFO:shinysdr:_SiteWithHeaders starting on 8100
> INFO:shinysdr:Starting factory <shinysdr.i.network.app._SiteWithHeaders instance at 0x112856248>
> INFO:shinysdr:OurStreamProtocol (FactoryWithArgs) (WebSocketFactory) starting on 8101
> INFO:shinysdr:Starting factory <shinysdr.twisted_ext.FactoryWithArgs instance at 0x1128561b8>
> INFO:shinysdr:Starting factory <txws.WebSocketFactory instance at 0x112856200>
> INFO:shinysdr:ShinySDR is ready.
> INFO:shinysdr:Visit http://localhost:8100/XhkB0cdV1ESfeoYqLknGqA

Also here's a screenshot: http://imgur.com/a/RxBwG

kpreid, 2017-02-19:
The startup info is all normal. Your screenshot link goes to an empty album so there's nothing for me to look at.

It may be that you simply have no receivers yet. To create one, click in the spectrum or on a frequency DB entry.

mehdideveloper, 2017-02-20:
@kpreid Thanks for the reply. 
I uploaded the screenshot again:  http://pasteboard.co/Av3ADBvTG.png
I see signals and I click on the spectrum and change them. It detects my RTL-SDR and shows the spectrum correctly, but there are no options for changing modulation and other related settings.

My system info:
```
OS X 10.12.2
Chrome 52
GNURadio 3.7.10.1
shinysdr (latest version fetched from git)
```

Also, this may help you:

> ...
> INFO:shinysdr:Unknown mode u'WFM' in Receiver(); using AM
> ....
> File "shinysdr/i/network/export_http.py", line 134, in render_POST
>     key = block.create_child(reqjson)  # note may fail
>   File "shinysdr/i/top.py", line 52, in create_child
>     (key, receiver) = self.__top.add_receiver(desc['mode'])
>   File "shinysdr/i/top.py", line 164, in add_receiver
>     ), state=combined_state)
>   File "shinysdr/values.py", line 660, in unserialize_exported_state
>     obj = ctor(**all_kwargs)
>   File "shinysdr/i/receiver.py", line 105, in __init__
>     self.__demodulator = self.__make_demodulator(mode, {})
>   File "shinysdr/i/receiver.py", line 386, in __make_demodulator
>     raise ValueError('Unknown mode: ' + mode)
> exceptions.ValueError: Unknown mode: AM

kpreid, 2017-02-20:
Aha, this means that `shinysdr.plugins.basic_demod` failed to load or there's something else seriously wrong. Scroll up and look for the _first_ traceback in the log — there should be some error other than "Unknown mode".

mehdideveloper, 2017-02-20:
Unfortunately there's no error other than "Unknown mode"
I have pasted the whole output log here:   http://pastebin.com/HwSg3S1h

kpreid, 2017-02-20:
Wow, okay, that's weird. I see from the HTTP log entries that other plugins (e.g. APRS) have been loaded (but `basic_demod` won't show up there because it doesn't load any client code), and there aren't any other errors.

Could you show the contents of http://localhost:8100/client/plugin-index.json ? That will show the list of modes that have been successfully loaded, so we can see if the problem is with the lookup somehow or if basic_demod is uniquely broken.

mehdideveloper, 2017-02-20:
Here's the contents:

`{"css":[],"js":["/client/plugins/shinysdr.plugins.aprs/aprs.js","/client/plugins/shinysdr.plugins.vor/vor.js","/client/plugins/shinysdr.plugins.mode_s/mode_s.js","/client/plugins/shinysdr.plugins.hamlib/hamlib.js"],"modes":{"VOR":{"can_transmit":true,"info_enum_row":{"description":null,"label":"VOR","sort_key":null,"type":"EnumRow"}}}}`

kpreid, 2017-02-20:
Okay, so all the usual plugins are being loaded but `basic_demod` isn't…

This continues to be weird. The only thing I can think of is to start troubleshooting in the actual code:

* Try removing all files named `dropin.cache` under `shinysdr/plugins/` (`find . -name dropin.cache -delete`). This shouldn't matter unless there was a previous version of the code in use which was broken in some way.
* Is `shinysdr/plugins/basic_demod.py` simply missing for some reason?
* Is the code in it getting executed (add print/logging to check)?
* `shinysdr.i.modes._ModeTable` is responsible for actually making the list of modes. Since `basic_demod` doesn't declare any of its modes conditionally-available I don't expect to see anything here but it's a place to look at what's going on.

mehdideveloper, 2017-02-20:
@kpreid Thank you very much, you made my day :)
Deleting the "dropin.cache" files did it.

closed 2017-02-20 by mehdideveloper

kpreid, 2017-02-20:
Oh, good. I'll have to add that to the troubleshooting guide when I get around to writing it.

Of course, it'd also be nice if it didn't silently break…