kpreid/shinysdr #60: Measviz.css and measviz.js Missing

aldude999, 2017-02-24:
I just installed shinysdr from scratch and when I connect a client to the server, it stalls at "Loading client...."

When checking the console, I get "Failed to load resource: the server responded with a status of 404" for http://ip-address:8100/client/measviz.css and
http://ip-address:8100/client/measviz.js

I've run fetch-js-deps.sh and setup.py install and created a folder and this is with a fresh install of Debian as well. the files exist in the deps folder too.

kpreid, 2017-02-25:
The list of files to install was not complete. Should be fixed in 240a399.

aldude999, 2017-02-26:
Cool, it's working! Now if only I could figure out how to get gr-dsd working properly. :P

closed 2017-02-26 by kpreid