kpreid/shinysdr #61: Timing out client: IPv4Address

bogdanr, 2017-02-26:
When trying to access it from the local network I get a timeout error:
```
INFO:shinysdr:Timing out client: IPv4Address(TCP, '192.168.1.141', 59722)
INFO:shinysdr:Timing out client: IPv4Address(TCP, '192.168.1.141', 59720)
```
Nevertheless, I can see the help page from the local network and from localhost I can access the application.
```
INFO:shinysdr:"192.168.1.141" - - [26/Feb/2017:21:23:01 +0000] "GET /manual/operation HTTP/1.1" 200 5353 "-" "Mozilla/5.0 (X11; Linux x86_64; rv:50.0) Gecko/20100101 Firefox/50.0"
INFO:shinysdr:"192.168.1.141" - - [26/Feb/2017:21:23:01 +0000] "GET /manual/manual-style.css HTTP/1.1" 304 - "-" "Mozilla/5.0 (X11; Linux x86_64; rv:50.0) Gecko/20100101 Firefox/50.0"
```

kpreid, 2017-02-26:
The `Timing out client` message means that the server did not see the client finish sending its request. What do you see from the browser?

bogdanr, 2017-03-01:
The browser also times out.

kpreid, 2017-03-02:
I'm afraid I can't make sense of what might be going on here, then. Do you perhaps have a web proxy or firewall of some sort in use on the browser you are trying to connect from?

You could try another web server instead of ShinySDR to see if you get the same result, such as `python -m SimpleHTTPServer` — running that command will start a web server serving just the files in the current directory.

bogdanr, 2017-03-02:
I most likely had a typo somewhere because now it works. Thanks for the help!

closed 2017-03-02 by bogdanr