kpreid/shinysdr #62: RTL SDR v3 direct sampling

vchav73, 2017-02-28:
First of all, I would like to say thanks for such a great tool! I tried quite a number of other spectrum analyzers before finding Shiny and am quite happy with it. 

I have a v3 RTL RDR with direct sampling support: http://www.rtl-sdr.com/rtl-sdr-blog-v-3-dongles-user-guide/. It seems Shiny does not support the direct sampling capability. Is that correct? If not, how could I go about enabling it?

kpreid, 2017-03-01:
I don't have any direct-sampling modified dongles to test with, unfortunately.

But ShinySDR uses gr-osmosdr to connect to RTL-SDR devices, so it's for the most part up to gr-osmosdr to support it. Based on checking the gr-osmosdr source code, you should be able to change your config from `OsmoSDRDevice('')` to `OsmoSDRDevice('rtl=0,direct_samp=2')` (note that the first part is just specifying to use an RTL-SDR device rather than any available device) to enable the direct sampling mode.

It sounds like the output also needs to be interpreted differently (based on that article's reference to "Q branch") but it will probably work with just a misaligned frequency scale and mirrored/duplicated spectrum. Try it out and let me know what happens, and I can look into adjusting the code.

vchav73, 2017-03-01:
Changing the config worked! I'm able to receive broadcast AM now. The frequency scale is even correct.

Can I configure direct sampling as a separate RF source or will I need to change the config and restart the server when I want to switch back and forth between direct sampling?

kpreid, 2017-03-02:
Unfortunately, that can only be set in the device string, not while the source is in operation (via gr-osmosdr; librtlsdr may be more flexible). The good news is, there's already support for dropping and rebuilding the source block — but it's there for TX/RX switching (no, there is no usable TX support at all, it's just a planned feature), so doing it to change options instead will need some new code. It shouldn't be hard. In fact just doing this should work:

    class _OsmoSDRRXDriver(...):
        ...
        def change_device(self, new_device_string):
            self.__osmo_device = new_device_string
            self._stop_rx()
            self._start_rx()

The remaining question would be how to present it in the UI:

* allowing picking from a list of device strings would be most flexible and generic, but would be bothersome to write out in the config file and not the best UI
* having a `direct_sampling` checkbox would be most user-friendly, but would be RTL-SDR-specific code.

vchav73, 2017-03-02:
Thanks for the info. Pretty cool to hear you're planning on Tx support.

I'll look at the code and see if I can make the changes (adding a checkbox) myself. Although I'm a developer, I have pretty much zero Python experience. It's a good opportunity to learn it and I'd be happy to contribute.