kpreid/shinysdr #63: cannot import name analog

demandzm, 2017-03-03:
I am getting this error message and cannot find what the problem is, any help would be great.

INFO:shinysdr:Log opened.
CRITICAL:shinysdr:main function encountered error
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 51, in main
    react(go)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-17.1.0-py2.7-linux-armv7l.egg/twisted/internet/task.py", line 908, in react
    finished = main(_reactor, *argv)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 46, in go
    return _main_async(reactor, argv, _abort_for_test)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-17.1.0-py2.7-linux-armv7l.egg/twisted/internet/defer.py", line 1447, in unwindGenerator
    return _inlineCallbacks(None, gen, Deferred())
--- <exception caught here> ---
  File "/usr/local/lib/python2.7/dist-packages/Twisted-17.1.0-py2.7-linux-armv7l.egg/twisted/internet/defer.py", line 1301, in _inlineCallbacks
    result = g.send(result)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 98, in _main_async
    execute_config(config_obj, args.config_path)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/config.py", line 263, in execute_config
    execfile(os.path.join(config_file_or_directory, 'config.py'), env)
  File "/home/chip/shinysdrconfig/config.py", line 7, in <module>
    from shinysdr.plugins.simulate import SimulatedDevice
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/simulate.py", line 24, in <module>
    from gnuradio import analog
exceptions.ImportError: cannot import name analog

INFO:shinysdr:Main loop terminated.

kpreid, 2017-03-04:
The module `gnuradio.analog` should be installed with GNU Radio. You have an incomplete GNU Radio installation.

demandzm, 2017-03-05:
I knew it was something simple but I just wasn't seeing it. As it turns out I was missing a dependency and gr-analog was disabled when I compiled gnu radio. However it seems this was a waste of time since it seems the C.H.I.P ($9 single board computer) doesn't have enough power to run everything properly. Thanks for the help anyway.

closed 2017-03-05 by kpreid