kpreid/shinysdr #64: Error opening database directory

mehdideveloper, 2017-03-12:
Hi @kpreid 
Whenever I install shinysdr for the first time on a system, it gives this error in the log:

> INFO:shinysdr:dbs/: Error opening database directory: OSError(2, 'No such file or directory')

I have to copy the folder "dbs" from "./shinysdr/data" to "." (root folder of shinysdr) to solve the problem

kpreid, 2017-03-12:
I am confused about why it is opening 'dbs/'. The default config-relative directory for databases is `dbs-read-only/`. What version (git commit id) of ShinySDR is this? What are the contents of your config file?

mehdideveloper, 2017-03-13:
I was using the "240a39970f10f3c47e8762f610587f7834acf4f2" (the latest version before your yesterday's commits)
Oh, I found what's wrong! I have this line in my config which is obviously what I'm experiencing:

`config.databases.add_directory('dbs/')`

Thank you @kpreid

closed 2017-03-13 by mehdideveloper

kpreid, 2017-03-13:
Thanks for the followup. That line used to be in the default config before I switched to the 'config directory' scheme, and in fact everyone would get that message. The intent was that you would make that directory if you need it, but I admit 'IOError' isn't a particularly friendly thing to say there.

Now, `dbs-read-only/` is checked only if it exists, and also created automatically as a hint that you can put things there.