kpreid/shinysdr #65: Input amplitude too high

mehdideveloper, 2017-03-12:
Dear @kpreid 
I have set up a shinysdr on an Ubuntu 16.04 x64 (with 2GB of RAM) and using SDRPlay as the receiver.
My client is the latest Chrome on Mac OS X.
On "Radio Config" dialog I see "Input amplitude too high. Reduce gain"
No matter how low or high I set the gain (or set it to auto), it always shows this.
Any ideas?

kpreid, 2017-03-12:
ShinySDR displays this message whenever the magnitude of any input sample exceeds 1.0. In most cases, the samples are originally fixed-point values and are rescaled into floats in a ±1 range, so a magnitude between 1.0 and 1.41 (√2, the maximum possible magnitude, the corner of the ±1 square) indicates the signal must be getting clipped.

* The message actually reads __`Input amplitude too high (X.XX ≥ 1.0).` What's the number you see there?__ If it is higher than 1.41 then it must be that the SDRPlay driver is reporting its output on a different scale and I will have to add a special case. But the source code doesn't seem to indicate that.

* Do you see any large spikes on the spectrum view? If you remove the antenna from the receiver does the message go away?

mehdideveloper, 2017-03-12:
The number is not constant. It changes. For example it goes as high as 17.00
Sometimes I see red spikes on the spectrum.
Removing the antenna almost fixes the value (around 16.00)

kpreid, 2017-03-12:
Okay, so it sounds like the SDRPlay driver is simply generating large numbers and we need to compensate for that. In that case, you can ignore the message for now. You may also go to the Options menu in the spectrum view and change the **Lowest value** and **Highest value** for whatever makes for comfortable viewing.

kpreid, 2017-03-12:
Possible solutions:
* Add some kind of metadata for the rx_driver to report its expected numeric range.
* Have `OsmoSDRRXDriver` rescale the input when a SDRPlay device is in use.
* Disable the clip warning if a value exceeding (√2 + ε) is ever seen (because that indicates the premise is faulty). No special cases, but also a quiet failure.

kpreid, 2017-09-23:
044a493da593c17bbd78377d25d1e0f9994454f6 implements the third option above: disable the clip warning if a value exceeding (√2 + ε) is ever seen.

I think that is enough for this issue.

closed 2017-09-23 by kpreid