kpreid/shinysdr #66: Sample rate change breaks waterfall

mehdideveloper, 2017-03-12:
Hi @kpreid 
I'm using SDRPlay with shinysdr with the sample rate set to 2000000 (2M)
I need to reduce it, so I set it to a lower valid value (like 600000) and then run shinysdr again and now I don't have a waterfall!
I stop the server, change the sample rate back to 2M and it's OK !
Is there any sort of cache that I need to delete to be able to use the new sample rate values?

Thanks

kpreid, 2017-03-12:
According to the source code for gr-osmosdr's SDRPlay driver, the recommended sample rates are between `2000e3` and `12000e3`. So your lower values are out of range.

(I'm aware that it would be useful if ShinySDR had some way of giving you this information, but I haven't thought of a good way to incorporate it yet. Changing the sample rate while running is reasonable but problematic for non-essential reasons.)

Why do you want to reduce the sample rate?

mehdideveloper, 2017-03-12:
Thanks I hadn't noticed that; So I should work on gr-osmosdr.
Well, I thought maybe running with a lower sample rate would give me both a faster experience (less bandwidth used) and also less CPU usage.
Am I missing something here?

kpreid, 2017-03-12:
Reducing the RF sample rate will significantly reduce CPU usage, but does not at all affect the network data rate; that is determined only by the waterfall settings and audio stream.

You can reduce the network data rate by keeping the waterfall rate and resolution settings low, and by turning off stereo audio (`config.features.disable('stereo')`). I'm thinking to add a control to reduce the audio stream sample rate, but right now that is entirely automatically chosen. You can reduce the value of `ASSUMED_USEFUL_SAMPLE_RATE` in `shinysdr/i/webstatic/client/audio.js` if you would like to try anyway.

mehdideveloper, 2017-03-13:
I don't know whether this is related to shinysdr, but for your information:
I changed the line in gr-osmosdr to allow setting sample rate from 250K to 10M and recompiled and reinstalled it on my system, but shinysdr still does not accept any sample rate below 2MSPS.
I know SDRPlay can accept lower sample rates, because first its developers have confirmed it and second I use it on SoapySDR based apps (like CubicSDR) with lower sample rates like 250KSPS.

Regards

kpreid, 2017-03-14:
I'm guessing this is a problem with gr-osmosdr, because there's just no reason ShinySDR should care about the specific sample rate, particularly if it is lower than one that worked. (The automatic sample rate conversion in ShinySDR has been tested across a lot of different rates and the worst it should ever do is run an overly CPU-expensive filter.)

The code in gr-osmosdr's `sdrplay_source_c::set_sample_rate` looks suspicious; it's failing to check the error code return and so that might be out of range as the diff-or-reinit test isn't really what the API documentation says. So I'd add a check of the return code and see what that says, and then maybe make it always reinit. But, again, I don't have a SDRPlay and can't do those experiments myself.

Good luck!

closed 2017-03-14 by kpreid