kpreid/shinysdr #67: Dropping data on LSB

mehdideveloper, 2017-03-12:
@kpreid Sorry for creating too many issues at once. I'm testing shinysdr (in fact stress-testing :) ) and am finding various errors.
A strange error is that when I switch to LSB modulation (on 40m ham band), I get this error:

> CRITICAL:shinysdr:u'Dropping data going to stream /ep2c/audio?rate=48000'

But I don't get this error in AM or FM Broadcast mod. Any hints?

kpreid, 2017-03-12:
In _theory_ this message indicates only that the server-to-client network connection isn't keeping up with the audio rate. As a result, it can occur whenever there's some kind of hiccup, including during a mode switch.

Do you see the message once, or continually? Does the audio actually sound OK?

If the audio is also stuttering then this is possibly related to #51.

mehdideveloper, 2017-03-12:
It just occurs with my Ubuntu machine ( an old machine) and only on SSB.
But not on my MacBook Pro.

Question: can I expect a faster shinysdr experience, if I run "volk_profile" on my machine?

kpreid, 2017-03-12:
Yes, you should run `volk_profile` in any case for efficiency and definitely if you're having any performance issues.

Please answer my questions: Do you see the message once, or continually? Does the audio actually sound OK?

mehdideveloper, 2017-03-12:
No it's not printed once; the message is repeatedly printed on the output log.
And the sound is not OK. It's like disconnects and connects

kpreid, 2017-03-12:
OK, so this is either #51 (that seems unlikely) or it's simply exceeding the capabilities of your old machine. Definitely do a `volk_profile`. Other than that, this simply falls under the category of improving overall CPU efficiency in ShinySDR (tracked in issue #17).

closed 2017-03-15 by kpreid