kpreid/shinysdr #68: Bandwidth compression

mehdideveloper, 2017-03-13:
Hi
Do you support compression of audio and other information sent from server to client?
How can it be compressed to make it usable on lower speed connections?

kpreid, 2017-03-14:
Audio compression is covered under #11. Spectrum data compression is also desirable but I have no idea how to go about it without distorting the displayed data. Everything else is insignificant comparatively.

closed 2017-03-14 by kpreid

kpreid, 2017-03-17:
You might be interested to know that I just pushed commit 06c0026856ba21c7a522bf1fb654d760c6188bd8 which does not add _compression_ but it does add a menu to easily choose a lower audio sample rate.

mehdideveloper, 2017-03-17:
Thanks. I was using a smaller hard-coded value. Now it's easier.