kpreid/shinysdr #71: GET /client/measviz.js HTTP/1.1" 404 - Measvis not loading

nlovlyn, 2017-04-15:
I am unable to get the webclient to load. It appears to be having issues with loading Measvis.I tried installing Measvis manually but it had no effect. In chrome the loading page will show up and the loading bar moves to about 2% but then it stops there. After about a minute the client times-out. I've tried to install this on Ubuntu 16.04 and 17.04. I am using the most current code and the latest version of chrome.

~$ sudo shinysdr ~/sdr
INFO:shinysdr:Log opened.
[INFO] [UHD] linux; GNU C++ version 6.3.0 20170406; Boost_106200; UHD_3.11.0.git-162-g2790b51f
gr-osmosdr v0.1.4-86-ge9dde9af (0.1.5git) gnuradio 3.7.12git-80-g3a2c67c6
built-in source types: file fcd rtl_tcp uhd rfspace redpitaya 
[INFO] [B200] Detected Device: B200mini
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] pass
[INFO] [AD936X] Performing CODEC loopback test... 
[INFO] [AD936X] CODEC loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [CORES] Performing timer loopback test... 
[INFO] [CORES] Timer loopback test passed
-- Using subdev spec 'A:A'.
[INFO] [B200] Asking for clock rate 36.571424 MHz... 
[INFO] [B200] Actually got clock rate 36.571424 MHz.
[INFO] [CORES] Performing timer loopback test... 
[INFO] [CORES] Timer loopback test passed
[WARNING] [MULTI_USRP] Setting DC offset is not possible on this device.
[WARNING] [MULTI_USRP] Setting IQ balance is not possible on this device.
100000000.0
gr::log :INFO: audio source - Audio source arch: alsa
INFO:shinysdr:Constructing...
INFO:shinysdr:Flow graph: Switching RF device to audio
INFO:shinysdr:Flow graph: Rebuilding connections because: initialization, switched device
INFO:shinysdr:Flow graph: ...done reconnecting (11 ms).
INFO:shinysdr:Restoring state...
INFO:shinysdr:Warning: Discarding nonexistent state <shinysdr.i.session.AppRoot object at 0x7fac6b2aeed0>.source_name = osmo 
INFO:shinysdr:Starting web server...
INFO:shinysdr:_SiteWithHeaders starting on 9102
INFO:shinysdr:Starting factory <shinysdr.i.network.app._SiteWithHeaders instance at 0x7fac6b22b8c0>
INFO:shinysdr:OurStreamProtocol (FactoryWithArgs) (WebSocketFactory) starting on 9103
INFO:shinysdr:Starting factory <shinysdr.twisted_ext.FactoryWithArgs instance at 0x7fac6b22bd88>
INFO:shinysdr:Starting factory <txws.WebSocketFactory instance at 0x7fac6b22bdd0>
INFO:shinysdr:ShinySDR is ready.
INFO:shinysdr:Visit http://localhost:9102/xrFFJor-tAJhmRCFYDVC_g/
INFO:shinysdr:"127.0.0.1" - - [15/Apr/2017:06:56:32 +0000] "GET /xrFFJor-tAJhmRCFYDVC_g/ HTTP/1.1" 200 4129 "-" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/57.0.2987.133 Safari/537.36"
INFO:shinysdr:"127.0.0.1" - - [15/Apr/2017:06:56:32 +0000] "GET /client/measviz.css HTTP/1.1" 404 145 "-" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/57.0.2987.133 Safari/537.36"
INFO:shinysdr:"127.0.0.1" - - [15/Apr/2017:06:56:32 +0000] "GET /client/plugin-index.json HTTP/1.1" 200 1410 "-" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/57.0.2987.133 Safari/537.36"
INFO:shinysdr:"127.0.0.1" - - [15/Apr/2017:06:56:32 +0000] "GET /client/measviz.js HTTP/1.1" 404 145 "-" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/57.0.2987.133 Safari/537.36"
INFO:shinysdr:Timing out client: IPv4Address(TCP, '127.0.0.1', 39306)
INFO:shinysdr:Timing out client: IPv4Address(TCP, '127.0.0.1', 39304)

kpreid, 2017-04-15:
You must run `fetch-js-deps.sh` before `setup.py install`; otherwise the Measviz files won't be copied into the installation location.

Also, I strongly recommend not running ShinySDR as root. It does not need to be root for any reason.

nlovlyn, 2017-04-15:
Thank you so much!!! I was doing that in the reverse order. It works now!

closed 2017-04-15 by nlovlyn