kpreid/shinysdr #74: Aliasing in waterfall can hide stations

bitglue, 2017-05-26:
If the waterfall resolution is high, and there's a narrow-band signal (like CW), then the fine tuning of the receiver can cause stations to disappear. Here's how I reproduce it:

1. Tune to an HF ham band with CW activity
2. Turn the waterfall resolution all the way up to 4096.
3. Make the browser window smallish / open the map, telemetry, frequency db etc windows to make the waterfall narrower
4. Adjust the radio tuning in 10 Hz steps.

The waterfall will appear to "shimmer", with stations alternating between strongly visible and invisible depending on the fine tuning. It seems to go away if the waterfall is made bigger by increasing the browser window size or closing other UI elements, so I have to guess it's aliasing is sampling the waterfall down to a smaller size.

kpreid, 2017-05-26:
Yep, this is a known issue. Part of the problem is that it's tricky to handle in a way which is both accurate when the horizontal size is small and efficient when the horizontal size is large (because whatever filter I use is compiled into the shaders and applied per-pixel).

It would be possible to cache the resampled spectrum in a second texture, so the cost would be paid per-new-line or at resizing rather than entire-viewport-per-frame, but I haven't implemented any piece of that.

The Official Workaround at the moment is to turn down the resolution control!

kpreid, 2017-05-26:
Also, the view-that-is-not-a-waterfall (why don't we have an unambiguous name for that yet) already has some minimal unscientific filtering (it tries to take samples across the span of one pixel, but not the right number of samples), whereas the waterfall has none.