kpreid/shinysdr #75: RTTY demod doesn't work

bitglue, 2017-05-27:
Given a very clean signal (like one I generate with FLDIGI) it _almost_ works, but still with some errors. :disappointed: 

I suspect this is because gr-rtty accumulates rounding error in its timing as it samples each symbol.

https://github.com/bistromath/gr-rtty/blob/master/lib/rtty_decode_ff_impl.cc#L109

More investigation to be done for sure. Wanted to attach some work in progress for others to follow along.

Included here is a GRC chart I've been using to play around. The example that comes with gr-rtty uses a quadrature demod, but from what I've read, that approach performs poorly in the presence of selective fading and isn't really used these days.

So I was playing with what I think kids these days are calling "AM detection", really just two filters tuned to each of the mark and space frequencies. Apparently this is how fldigi does it, and it references this neat paper on threshold detection. Seems like the threshold detection is where the non-obvious design decisions are made.

http://www.w1hkj.com/FldigiHelp-3.21/html/rtty_page.html
http://www.w7ay.net/site/Technical/ATC/

Also included are some recordings. They are the usual 45.45 baud RTTY, 170 Hz spacing, centered on 2kHz audio. There are a couple calling CQ, one with "RYRYRY..." (alternating marks and spaces), and one with "empty" diddles.

[rtty.zip](https://github.com/kpreid/shinysdr/files/1033215/rtty.zip)

Will update this issue later with progress over the weekend, with any luck.

bitglue, 2017-05-27:
TIL: Sources are conflicted about the number of conventional stop bits: either 2 or 1.5. FLDIGI generates 1.5. [gr-rtty](https://github.com/bistromath/gr-rtty/blob/master/lib/rtty_decode_ff_impl.cc#L64) talks about "old-school" 1.42 stop bits. [Wikipedia says](https://en.wikipedia.org/wiki/Teleprinter#Teleprinter_operation) "Since modern computer equipment cannot easily generate 1.42 bits for the stop period, common practice is to either approximate this with 1.5 bits, or to send 2.0 bits while accepting 1.0 bits receiving." Which is odd, because I thought modern computing had caught up to 1940s electromechanical equipment by now.

I guess what they mean to say is a synchronous protocol can't have 1.42 stop bits, but then it can't have 1.5, either. So I think the lesson is any attempt at synchronous clock recovery is incorrect since in practice we could receive anything between 1.42 and 2 stop bits.

kpreid, 2017-05-27:
Seems to me one could use "half-bits" in the clock recovery (I tried to do that in [my scratch version](https://github.com/kpreid/shinysdr-morgue/blob/rtty-demodulator-from-scratch/shinysdr/plugins/rtty/__init__.py)), and have the synchronization be fast enough to accept the difference between 1.42 and 1.5.

bitglue, 2017-05-28:
I have RTTY demod working in GRC. Wrote my own blocks because gr-rtty just wasn't cutting it. Would appreciate a code review: https://github.com/bitglue/gr-baudot

I've some research and experimentation to do on the filtering for the detectors. The chart I hacked together just uses whatever filtering was easiest to type, and probably isn't great in the presence of noise. Then it's "just" a matter of integrating it.

PR #76 (2017-06-01): Switch RTTY to gr-radioteletype
gr-rtty didn't work. gr-radioteletype is available at:

    https://github.com/bitglue/gr-radioteletype

Fixes #75.
commit 68e5173f (2017-06-01): Switch RTTY to gr-radioteletype

gr-rtty didn't work. gr-radioteletype is available at:

    https://github.com/bitglue/gr-radioteletype

Fixes #75.
closed 2017-06-01 by kpreid