kpreid/shinysdr #83: Map zoom very slow

bitglue, 2017-06-07:
To go from seeing the entire globe to zooming in on Michigan takes me 17 full pushes of the scroll wheel. Maybe it could be faster?

kpreid, 2017-06-07:
The tradeoff of setting the step size larger is not being able to set the map viewport where you put it, and I was wary of making things too fast because scroll-wheel behavior can vary across platforms. (For example, it takes only five for me.)

On consideration of your feedback and how it feels on my system, I've now multiplied the zoom rate by 10 in e06f23e3b8a768678ee6eecb1a38bc35f63e8ecc. Let me know how it feels.

I also pushed a bug fix that had gotten buried in another project: the map now (again) has numerical lat/long/zoom fields.

bitglue, 2017-06-07:
Much better, yes. I'm using Google Maps as a point of reference, and it seems the zoom speed is about the same now. Thanks again!

bitglue, 2017-06-07:
Unrelated, but at least for me the numerical display is truncated in a rather ugly way:

[image]

closed 2017-06-07 by bitglue

kpreid, 2017-06-07:
> I'm using Google Maps as a point of reference, and it seems the zoom speed is about the same now.

Google Maps has acceleration, though, so it's not quite directly comparable.

> the numerical display is truncated in a rather ugly way

Yeah, known issue. Need to truncate to a specific number of digits and make sure that this doesn't quantize the actual pan/zoom levels. It's not a priority compared to lots of other stuff.