kpreid/shinysdr #84: UI notes

bitglue, 2017-06-07:
I've been thinking I should keep a log of UI things that have confused or surprised me. No particular action required, just things to consider.

- Sometimes I try to drag the waterfall left or right to pan it, and end up creating or moving a receiver instead.
- Sometimes I get tired of horizontally scrolling if I have a long distance to go. This might be why I sometimes drag to pan. Also, not everyone has a horizontal scroll.
- It took me a while to figure out I can mouse-wheel the numbers on the frequency display.
- It bugs me that clicking on the 20m band in the frequency database always takes me to the middle of the band, even though it's always the lower part of it I'm interested in seeing. Perhaps it can remember where in the band I left it? This is how my FT-897 works.
- If I have several receivers open, I try to scroll through them with the mouse wheel, which works until a frequency widget is under the pointer. Then it ends up tuning.
- When there are multiple receivers, I have no idea which one will be changed if I click on the waterfall.
- When I create a new receiver, it's always AM which is never what I want. I'm not sure exactly what I'm expecting, but it's not AM. Maybe same as the last thing? Maybe it's confusing because clicking the waterfall sometimes makes a new receiver, and other times it changes an existing one? Maybe the default can be based on the band defined in the frequency database?
- I couldn't find a way to associate receivers on the waterfall with receivers in the radio config column until just now, when I double-checked to make sure there wasn't a way. The little letter at the top of the cursor isn't very obvious. And even so, it's tedious to look at the letter, then scan through the receivers to find the right one. Especially given scrolling difficulties previously mentioned.
- "Radio Config" says "RF Source: Audio" but I'm expecting "softrock" as defined in my config:

```
softrock_audio = AudioDevice(sample_rate=96000, channel_mapping='IQ', usable_bandwidth=(1000, 96e3/2-60))
position = PositionedDevice(42.60889, -83.54582)
d = hamlib.connect_to_rig(config.reactor, options=['-m', '2509', '-r', '/dev/softrock'])
d.addCallback(lambda rig: config.devices.add('softrock', rig, softrock_audio, position))
config.wait_for(d)
```

- It's tedious re-adjusting the range on the waterfall when switching bands. Perhaps this is another setting that can be remembered per-band?
- The menu under the hamburger doesn't seem to quite fit typical patterns. Looking on Google maps (android and web), and Google Play, and Slack (the three things I have open that have hamburgers), the hamburger should open a menu on the left: selecting something in the menu takes me to a new screen and closes the menu.

bitglue, 2017-06-07:
* The Audio spectrum is paused by default. Usually I just think it's broken. Just now it took me 30 seconds of head scratching to figure out how I made it work last time.

bitglue, 2017-06-07:
* If I'm zoomed in on the waterfall, and I pick a frequency from the DB, sometimes the new receiver ends up in a part of the waterfall which isn't currently visible and it takes me a moment to figure out why it disappeared.

bitglue, 2017-06-08:
* shift-clicking on the frequency database creates a new receiver, but also selects a bunch of text.

bitglue, 2017-06-08:
* I've thought maybe clicking on a receiver which is inactive would tune the radio so it can be active. But I've not found a better way than manually tuning.

kpreid, 2017-06-08:
Thanks for all the feedback.

> The Audio spectrum is paused by default. Usually I just think it's broken. Just now it took me 30 seconds of head scratching to figure out how I made it work last time.

This is because I don't yet have a good way to structure "run only if visible". Once I do, or if I make it more like "create the window" than "unhide the window" then it will not be paused by default.

> shift-clicking on the frequency database creates a new receiver, but also selects a bunch of text.

Fixed.

> "Radio Config" says "RF Source: Audio" but I'm expecting "softrock" as defined in my config

You can pass `name='Softrock'` or whatever when constructing the device, but now that you've pointed it out using the key as the name by default makes sense.

> It took me a while to figure out I can mouse-wheel the numbers on the frequency display.

That's why this is one of the few UI elements that is actually documented in the manual! Not sure how to improve discoverability of this.

> When I create a new receiver, it's always AM which is never what I want.  … Maybe the default can be based on the band defined in the frequency database?

This is already implemented, but the default frequency databases don't e.g. specify LSB/USB as whole-band defaults. I was wary of being overly-specific, but perhaps they should. This would imply also adding at least the legally-defined CW/data segments, if not an entire band plan (I haven't done _that_ because I don't want to exactly duplicate copyrighted text nor rewrite it inaccurately).

kpreid, 2017-07-30:
Commit 0c8b1019f552c82730493e585c4ddf8b4e5bae39 splits amateur bands and marks the lower half as CW and the upper half as LSB or USB as appropriate.