kpreid/shinysdr #91: Error running ShinySDR on git pull June-18-2017@12:29am UTC

techfixpros, 2017-06-29:
Ran a git pull today, re-ran the fetch-js-deps.sh, built and installed and upon running received the following:
`Traceback (most recent call last):
  File "/usr/local/bin/shinysdr", line 11, in <module>
    load_entry_point('ShinySDR==0.0.0', 'console_scripts', 'shinysdr')()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 561, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2649, in load_entry_point
    return ep.load()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2303, in load
    return self.resolve()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2309, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 31, in <module>
    from twisted.internet import reactor as singleton_reactor
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/reactor.py", line 38, in <module>
    from twisted.internet import default
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/default.py", line 56, in <module>
    install = _getInstallFunction(platform)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/default.py", line 44, in _getInstallFunction
    from twisted.internet.epollreactor import install
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/epollreactor.py", line 24, in <module>
    from twisted.internet import posixbase
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 18, in <module>
    from twisted.internet import error, udp, tcp
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 28, in <module>
    from twisted.internet._newtls import (
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/_newtls.py", line 21, in <module>
    from twisted.protocols.tls import TLSMemoryBIOFactory, TLSMemoryBIOProtocol
  File "/usr/local/lib/python2.7/dist-packages/twisted/protocols/tls.py", line 63, in <module>
    from twisted.internet._sslverify import _setAcceptableProtocols
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/_sslverify.py", line 38, in <module>
    TLSVersion.TLSv1_1: SSL.OP_NO_TLSv1_1,
AttributeError: 'module' object has no attribute 'OP_NO_TLSv1_1'
`
Deleted the complete shinysdr folder and ran a fresh git clone and then the usual steps to build/install and receive the same error. It had been working from a repo clone from a month ago until I attempted to upgrade today. I am unsure what versions/libs you may need to know.

Linux Mint 17.2

techfixpros, 2017-06-29:
Solved by downgrading  Twisted from 17.1.0 using: pip install Twisted==16.0.0. Then the server would run but with an error:
`:0: UserWarning: You do not have a working installation of the service_identity module: 'No module named cryptography.x509'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module and a recent enough pyOpenSSL to support it, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.`

This was then fixed by using: pip install cryptography

kpreid, 2017-06-29:
Upgrading exactly PyOpenSSL should be sufficient to fix this, as I just found out recently while unbreaking the [Travis CI build](https://travis-ci.org/kpreid/shinysdr).

Since this is an issue with version compatibility between dependencies, I don't think there's anything strictly to do here, but if other people have the same problem without fixing it themselves I'll add it to the dependencies checker and/or manual.

closed 2017-06-29 by kpreid