kpreid/shinysdr #92: Allow specifying root URLs separately from endpoints

kpreid, 2017-08-05:
Add a configuration option to specify the root URLs explicitly rather than deriving them from the `endpoint` options. This will enable using ShinySDR behind a reverse proxy, port forwarding, or other such configurations.

Relatedly, it would be nice to have a standard reverse proxy configuration (e.g. for nginx which I vaguely hear is suitable for this task) that makes ShinySDR use only one port rather than two.

thefinn93, 2017-11-13:
I would like to begin working on this, and have a few questions about how you would prefer to have it done.

It looks like the most reasonable place to specify it in the config would be in parameters of [`config.serve_web`](https://github.com/kpreid/shinysdr/blob/bc1a075d96db517b24481e9bd86d5aa2ef567d7c/shinysdr/config.py#L105), which would then get passed to [`shinysdr.i.network.app.WebService`](https://github.com/kpreid/shinysdr/blob/fdd17db4aac963a154a994478ddc0030a2e7dd39/shinysdr/i/network/app.py#L99), but there's a comment there indicating you feel there are already too many parameters. I could create setter methods to configure the websocket URIs there, but that feels rather awkward. What would you suggest? This is just based on me poking around and trying to understand the code, I dont really have a good understanding of the whole codebase, so there may be a better option...

kpreid, 2017-11-13:
Certainly this should be additional parameters to `config.serve_web`, as that makes sense from the config-file-design perspective. (I currently think a good name for them would be `base_(http|ws)_url`; not using the word "root" because it has another use.)

The “too many parameters” note on `WebService` is more "this is suspicious and should probably be refactored" than any specific desired upper limit. Feel free to just add it there too for now. Don't add setters for this reason; immutable things are to be preferred over mutable things.

Actually, I think it would make sense to have an object (namedtuple or otherwise) that pairs up the `endpoint` with the URL; then `endpoint_string_to_url` can also become a method of that object.

kpreid, 2020-04-19:
As of commit 15ba713375909146ba04a4448d0ec7106b662df5, there are two new parameters to `config.serve_web(...)`, `http_base_url` and `ws_base_url`, which allow overriding the URL generation. I have not yet tested this with an actual reverse proxy.

closed 2020-04-19 by kpreid

kpreid, 2020-04-19:
Belated followup: with f78ecfdcd9e4f65a982033230798fe647ab8b664 the ws_base_url is allowed to have a path, which is likely to be useful in reverse proxy configurations.