kpreid/shinysdr #96: CRITICAL error in i/network/export_ws.py - exceptions.UnicodeDecodeError: 'ascii' codec can't decode byte 0xae: ordinal not in range(128)

vpoluyaktov, 2017-10-03:
Just updated kpreid/shinysdr master branch, rebuilt all and got critical error (see log below).
The shinysdr server hangs after that.

ubuntu@h1:[git:upstream/master]~/git/shinysdr$ git remote -v | grep upstream
upstream        https://github.com/kpreid/shinysdr.git (fetch)
upstream        https://github.com/kpreid/shinysdr.git (push)

ubuntu@h1:[git:upstream/master]~/git/shinysdr$ git pull
Already up-to-date.
ubuntu@h1:[git:upstream/master]~/git/shinysdr$ git status
On branch upstream/master
Your branch is up-to-date with 'remotes/upstream/master'.
nothing to commit, working directory clean

./fetch-js-deps.sh
...skipped...
rm -rf build/*
...skipped...
python setup.py build
...skipped...
sudo python setup.py install
...skipped...
sudo rm -r /etc/shinysdr/state.json

ubuntu@h1:[git:upstream/master]~/git/shinysdr$ sudo shinysdr /etc/shinysdr/
INFO:shinysdr:Log opened.
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.9
built-in source types: file fcd rtl rtl_tcp rfspace soapy redpitaya
[WARNING] SoapySSDPEndpoint failed join group udp://[ff02::c]:1900
  setsockopt(IPV6_ADD_MEMBERSHIP) [19: No such device]
[INFO] Using format CF32.
100.0
INFO:shinysdr:Constructing...
Using Volk machine: avx2_64_mmx_orc
INFO:shinysdr:Flow graph: Switching RF device to osmo
INFO:shinysdr:Flow graph: Rebuilding connections because: initialization, switched device
INFO:shinysdr:Flow graph: ...done reconnecting (4 ms).
INFO:shinysdr:Restoring state...
INFO:shinysdr:Warning: Discarding nonexistent state <shinysdr.i.session.AppRoot object at 0x7ff1484d1fd0>.source_name = osmo
INFO:shinysdr:Starting web server...
INFO:shinysdr:_SiteWithHeaders starting on 48100
INFO:shinysdr:Starting factory <shinysdr.i.network.app._SiteWithHeaders instance at 0x7ff138bbe638>
INFO:shinysdr:OurStreamProtocol (FactoryWithArgs) (WebSocketFactory) starting on 48101
INFO:shinysdr:Starting factory <shinysdr.twisted_ext.FactoryWithArgs instance at 0x7ff1423c6ea8>
INFO:shinysdr:Starting factory <txws.WebSocketFactory instance at 0x7ff138bbe5f0>
INFO:shinysdr:ShinySDR is ready.
INFO:shinysdr:Visit http://localhost:48100/GnVMwPngOqqFw5RU-hjKSA/
INFO:shinysdr:Starting RFC 6455 conversation
INFO:shinysdr:Stream connection to  /GnVMwPngOqqFw5RU-hjKSA/radio
Poller starting
INFO:shinysdr:Starting RFC 6455 conversation
INFO:shinysdr:Stream connection to  /GnVMwPngOqqFw5RU-hjKSA/audio?rate=22050
INFO:shinysdr:Flow graph: Rebuilding connections because: added audio queue
INFO:shinysdr:Flow graph: ...done reconnecting (343 ms).
INFO:shinysdr:"64.236.138.2" - - [03/Oct/2017:17:50:20 +0000] "GET /GnVMwPngOqqFw5RU-hjKSA/ HTTP/1.1" 200 4209 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_12_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/60.0.3112.113 Safari/537
.36"
INFO:shinysdr:Closing connection: '' (1001)
INFO:shinysdr:Flow graph: Rebuilding connections because: removed audio queue
INFO:shinysdr:Flow graph: ...done reconnecting (362 ms).
INFO:shinysdr:Closing connection: '' (1001)
INFO:shinysdr:"64.236.138.2" - - [03/Oct/2017:17:50:21 +0000] "GET /favicon.ico HTTP/1.1" 200 1006 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_12_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/60.0.3112.113 Safari/537.36"
Poller stopping
INFO:shinysdr:"64.236.138.2" - - [03/Oct/2017:17:50:21 +0000] "GET /client/plugin-index.json HTTP/1.1" 200 2299 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_12_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/60.0.3112.113 Safari/53
7.36"
INFO:shinysdr:"64.236.138.2" - - [03/Oct/2017:17:50:21 +0000] "GET /GnVMwPngOqqFw5RU-hjKSA/wdb/ HTTP/1.1" 200 33 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_12_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/60.0.3112.113 Safari/5
37.36"
INFO:shinysdr:"64.236.138.2" - - [03/Oct/2017:17:50:21 +0000] "GET /GnVMwPngOqqFw5RU-hjKSA/dbs/ HTTP/1.1" 200 1689 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_12_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/60.0.3112.113 Safari
/537.36"
INFO:shinysdr:Starting RFC 6455 conversation
INFO:shinysdr:Stream connection to  /GnVMwPngOqqFw5RU-hjKSA/radio
[... 41 lines cut ...]
    result = inContext.theWork()
  File "/usr/local/lib/python2.7/dist-packages/Twisted-17.9.0-py2.7-linux-x86_64.egg/twisted/python/threadpool.py", line 266, in <lambda>
    inContext.theWork = lambda: context.call(ctx, func, *args, **kw)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-17.9.0-py2.7-linux-x86_64.egg/twisted/python/context.py", line 122, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-17.9.0-py2.7-linux-x86_64.egg/twisted/python/context.py", line 85, in callWithContext
    return func(*args,**kw)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/network/export_ws.py", line 326, in _AudioStream_read_loop
    buf += message.to_string()
exceptions.UnicodeDecodeError: 'ascii' codec can't decode byte 0xae in position 439: ordinal not in range(128)

INFO:shinysdr:Performing state write...
INFO:shinysdr:...done
^CINFO:shinysdr:Received SIGINT, shutting down.
INFO:shinysdr:(TCP Port 48100 Closed)
INFO:shinysdr:Stopping factory <shinysdr.i.network.app._SiteWithHeaders instance at 0x7ff138bbe638>
Poller stopping
INFO:shinysdr:(TCP Port 48101 Closed)
INFO:shinysdr:Stopping factory <shinysdr.twisted_ext.FactoryWithArgs instance at 0x7ff1423c6ea8>
INFO:shinysdr:Stopping factory <txws.WebSocketFactory instance at 0x7ff138bbe5f0>
INFO:shinysdr:Flow graph: Rebuilding connections because: removed audio queue
^CINFO:shinysdr:Received SIGINT, shutting down.
INFO:shinysdr:Flow graph: ...done reconnecting (371 ms).
^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C
`

kpreid, 2017-10-04:
Thanks for the report! Should be fixed now.

If you notice any other unicode-related errors, even if they're not apparently interfering with operation, please let me know. I recently moved to `from __future__ import unicode_literals` mode everywhere and there aren't enough tests to catch every forgotten `b''`.

closed 2017-10-04 by kpreid

vpoluyaktov, 2017-10-04:
Sure I’ll do. Thank you for quick fix!