kpreid/shinysdr #97: It's not possible to save DC cancellation state for soapy device

vpoluyaktov, 2017-10-12:
I use SDRPlay2 device as:
config.devices.add(u'osmo', OsmoSDRDevice('soapy', name='SDRPlay2'))

Found a problem with saving a state of DC cancellation checkbox on Radio Config widget.
By default DC cancelation is off and i see high peak in center frequency. 

When I click on the check box I can see how DC cancellation is turned On from the log:
INFO:shinysdr:set /XXXXXXXXXXXXXXX/radio/sources/osmo/rx_driver/dc_cancel to True (0.00s) 

Also the high peak in center frequency disappears. But DC cancellation checkbox returns to unchecked state immediately after the click. 

I believe that _set_dc_offset_mode(mode, ch)_  always returns False in my case (not sure about other devices). 

As result of this DC cancellation checkbox state is always false:
_self.dc_state = self.__source.set_dc_offset_mode(mode, ch)_
although DC cancellation mode switch is working.

kpreid, 2017-10-13:
Thanks for finding this. IQ balance handling was also wrong in the same way.

closed 2017-10-13 by kpreid

vpoluyaktov, 2017-10-13:
Thank you for quick fix!