kpreid/shinysdr #99: FR Spectrum and audio freeze often if Hi-Z SDRPlay2 antenna selected

vpoluyaktov, 2017-10-17:
Noticed a problem with SDRPlay2 (config.devices.add(u'osmo', OsmoSDRDevice('soapy', name='SDRPlay2')). When I use 'Antenna A' or 'Antenna B' shinnysdr is stable enough but if I switch to antenna 'Hi-Z' FR Spectrum and audio freeze very often. 
I do see nothing suspicious in the log, just regular stuff: 

INFO:shinysdr:set /GnVMwPngOqqFw5RU-hjKSA/radio/sources/osmo/rx_driver/gain to 46.4734848484848 (0.00s)
INFO:shinysdr:set /GnVMwPngOqqFw5RU-hjKSA/radio/sources/osmo/rx_driver/gain to 46 (0.00s)
INFO:shinysdr:set /GnVMwPngOqqFw5RU-hjKSA/radio/receivers/a/rec_freq to 2719565.217391304 (0.00s)
INFO:shinysdr:Scheduling state write.
INFO:shinysdr:Performing state write...
INFO:shinysdr:...done
INFO:shinysdr:set /GnVMwPngOqqFw5RU-hjKSA/radio/receivers/a/rec_freq to 3658695.652173913 (0.00s)
INFO:shinysdr:Scheduling state write.
INFO:shinysdr:Performing state write...
INFO:shinysdr:...done
INFO:shinysdr:set /GnVMwPngOqqFw5RU-hjKSA/radio/sources/osmo/rx_driver/antenna to u'Antenna A' (0.00s)
INFO:shinysdr:Scheduling state write.
INFO:shinysdr:Performing state write...
INFO:shinysdr:...done
INFO:shinysdr:set /GnVMwPngOqqFw5RU-hjKSA/radio/sources/osmo/rx_driver/antenna to u'Hi-Z' (0.40s)
INFO:shinysdr:Scheduling state write.
INFO:shinysdr:Performing state write...
INFO:shinysdr:...done

Not sure is it a SDRPlay2 problem or shinysdr one. 
I made public server so you can play with it http://h1.vp-seattle.us:48100/GnVMwPngOqqFw5RU-hjKSA/. Try to switch between antennas and reload the page. Sometimes you need some patience to make Hi-Z work it's working. Trust me.
Let me know when you are done - i'll change cap url.  

Thank you!

kpreid, 2017-10-17:
Your server does not seem to be up or reachable. It's unlikely that this can be adequately debugged solely from the UI, but some things to try:

* When it freezes do the controls remain responsive? In particular, can you create/delete receivers or switch modes successfully, they just have no output?

* If you delete all receivers and toggle the "pause" checkbox on the waterfall, does that make a difference?

* If you have any other hardware (e.g. an RTL-SDR), configure the server to make both available and see if switching to it and back makes the SDRPlay device recover.

----

The fact that _only_ setting the antenna value causes it to hang points strongly at the driver/hardware as being the problem, because ShinySDR does not do anything for `antenna` but pass on the command to `gr-osmosdr`.

I plan to eventually switch to a model where it is possible to reconnect to individual devices without restarting the server; for now, you can [enable the `reboot` feature](https://kpreid.github.io/shinysdr/manual/configuration) to allow restarting the server remotely, to make this problem less problematic.

vpoluyaktov, 2017-10-17:
Sorry, I was playing rebuilding it. Try now.

When it freezes do the controls remain responsive? In particular, can you create/delete receivers or switch modes successfully, they just have no output?
> Yes, I can do everything, just no output. The page refresh usually helps.

If you delete all receivers and toggle the "pause" checkbox on the waterfall, does that make a difference?
> Nope. I tried this. 

If you have any other hardware (e.g. an RTL-SDR), configure the server to make both available and see if switching to it and back makes the SDRPlay device recover.
> I enable Simulated RF device. Tried to switch back an force. The Simulated device is working, but SDRPlay2 is not.

kpreid, 2017-10-17:
Problems of this nature are _usually_ due to the GR block not coping well with ShinySDR's start/stop/reconfiguration (necessary due to supporting multiple receivers and clients and such). But it's still weird that it happens in this particular case and that way (particularly that reloading helps but other things don't).

I'll leave this open in case anyone else has some input, but I don't expect I'll be able to fix this for you, not having a SDRPlay2 device to test with.