kpreid/shinysdr #101: DSD issue

DanielCraig, 2018-01-11:
Hi,

I was having trouble getting gr-dsd to work, got it to show up in the web UI, but now received the following error when attempting to use "DSD".

'''
CRITICAL:shinysdr:Unhandled Error
Traceback (most recent call last):
Failure: exceptions.TypeError: Required argument 'frame' (pos 1) not found
'''

I was able to get the plugin functioning by investigating the import process in a python shell. I could only get "dsd_block_ff" to load on my setup by using "from dsd import dsd_block_ff". I made this same change to the dsd.py file in /plugins/, and the DSD option now appears in the web UI.

Clicking it, however, yields the above error message. Do you have any thoughts as to what the issue is?

kpreid, 2018-01-12:
This is almost certainly a change in gr-dsd's interface. Can you specify exactly which version you are using?

DanielCraig, 2018-01-12:
I used pybombs to install it, but the CHANGELOG file shows 1.7.0-dev as the latest entry.  It looks like the last commit to github.com/argilo/gr-dsd was a little over a year ago (Oct 28, 2016), as was the last update to the pybombs recipe file.

kpreid, 2018-01-12:
Unfortunately, I'm having trouble compiling that version of gr-dsd and getting a result that works at all. If you can look into this further yourself, please do (I'd start with looking for other usage examples and digging around in interactive python with `import dsd` and so on); otherwise I'll try again on another platform later.

DanielCraig, 2018-01-23:
Sorry for the delay in responding. I've confirmed that the dsd.py file has to be edited to "import dsd_block_ff" explicitly for the DSD mode to show up in the ShinySDR UI, but I am still getting the same error in the command line when I try to use it.
This was confirmed on a second installation of the software, where everything else seems to work properly (except for maybe the zoom in/out functionality)

kpreid, 2018-01-23:
Yes, the parameters have changed. I just took a look at the code on git and the new `dsd_block_ff` has the C++ signature

```
static sptr make(
    dsd_frame_mode frame,
    dsd_modulation_optimizations mod,
    int uvquality,
    bool errorbars,
    int verbosity);
```

so fixing this requires figuring out the correct values for those new parameters.

DanielCraig, 2018-01-23:
I've been looking at the DSD block in GRC, to see if I can get it working there first, and to get familiar with the settings. As it turns out, I ran across the response you gave [here](https://ham.stackexchange.com/questions/5844/sdr-gnuradio-gr-dsd-decode-d-star-voice).
I have a flowgraph that is receiving my DMR handheld, but I can't seem to get the decoding to work.
I'll keep poking at it and see what I can come up with.
Are you on any of the SDR IRC channels?

DanielCraig, 2018-01-24:
I think the place to start would be the default parameters used by the gr-dsd "DSD Block", as shown in the [screenshot](https://i.stack.imgur.com/Mfms2.png) from the stackexchange thread I linked above. I'll spend some time looking into how to set these values in ShinySDR, but please let me know your thoughts.

DanielCraig, 2018-01-24:
I seem to have the DSD plugin operating. I am still lacking actual decoding, in the same way that I am in GRC, but the plugin doesn't seem to be throwing any exceptions. For now it's just hardcoded too.
In dsd.py:
Changed import on line 29 to
```
"import dsd"
```
Changed dsd_block on line 69 to
```
"dsd.dsd_block_ff(dsd.dsd_FRAME_DMR_MOTOTRBO,dsd.dsd_MOD_GFSK,3,True,2),"
```

DanielCraig, 2018-01-25:
Do you have any suggestions for adding a menu to select the options? Perhaps similar to the dropdown menu that appears for "AM demodulation". The labels and values are as follows:

Dropdown 1:
Title:
"Frame Type"
Options:
Label: "Auto-detect" == value: "dsd.dsd_FRAME_AUTO_DETECT"  ----- This should be the default
Label: "P25 Phase 1" == value: "dsd.dsd_FRAME_P25_PHASE_1"
Label: "D-STAR" ==  value: "dsd.dsd_FRAME_DSTAR"
Label: "NXDN48 (6.25 kHz) / IDAS" == value: "dsd.dsd_FRAME_NXDN48_IDAS"
Label: "NXDN96 (12.5 kHz)" == value: "dsd.dsd_FRAME_NXDN96"
Label: "ProVoice" == value: "dsd.dsd_FRAME_PROVOICE"
Label: "DMR/MOTOTRBO" == value: "dsd.dsd_FRAME_DMR_MOTOTRBO"
Label: "X2-TDMA" == value: "dsd.dsd_FRAME_X2_TDMA"

Dropdown 2:
Title: "Mod Optimization"
Label: "Auto-select" == value: "dsd.dsd_MOD_AUTO_SELECT"  ----- This should be the default
Label: "C4FM" == value: "dsd.dsd_MOD_C4FM"
Label: "GFSK" == value: "dsd.dsd_MOD_GFSK"
Label: "QPSK" == value: "dsd.dsd_MOD_QPSK"

kpreid, 2018-01-26:
That's feasible. I am going to prepare a "blind" change to at least add reasonable defaults and possibly the menus.

kpreid, 2018-01-26:
@DanielCraig I just pushed a change that should implement the no-options version. Let me know if it works at all.

DanielCraig, 2018-01-26:
That seems to work, right-out-of-the-box.

```
+                uvquality=3,  # TODO figure out what this does
```

This is described on the [DSD Operation page](https://github.com/szechyjs/dsd/wiki/Operation) towards the bottom of the page.

**
Unvoiced speech is supposed to be generated with a noise source, 256 point DFT a number of band filters, followed by a 256 point inverse DFT. For computational simplicity mbelib uses a different method. For each unvoiced speech band, a number of sine waves are generated, each with a different random initial phase. The number of waves used per band is controlled by the -u option. A setting of 4 would approximate the performance of the 256 point DFT method as the maximum number of voice bands is 56, and very low frequencies are not synthesized. Values less than 3 have a noticable lack of unvoiced speech and/or artifacts. The defualt of 3 provides good speech quality with reasonable cpu use. Increasing the quality above the default rapidly consumes more CPU for increasingly diminishing returns.
**

kpreid, 2018-01-26:
Thanks. I was looking at source code and not separated docs.

Working on an update that includes a control for uvquality. The other options will require some way to get the enum values out of gr-dsd (hardcoding them is not acceptable).

DanielCraig, 2018-02-07:
Sounds good. Thank you.
Are you planning on adding the UI options to another Issue? Being able to select DSD options in the UI.

kpreid, 2018-02-08:
I already mentioned:

> The other options will require some way to get the enum values out of gr-dsd (hardcoding them is not acceptable).

DanielCraig, 2018-02-14:
I'm not sure I understand what you mean by enum values.

kpreid, 2018-02-14:
The values like `dsd.dsd_FRAME_DSTAR`. There needs to be a way to get the list of them out of gr-dsd rather than putting the list in ShinySDR's source.