kpreid/shinysdr #104: NOAA APT weather satellites

AlanJan, 2018-02-16:
Hi,

Is there a way to modify the VOR plugin to receive NOAA APT weather satellites?(or a other plugin)
I need to modify the filter bandwidth to 36Khz for the APT signal.
so  if the VOR plugin has been modified, must we recompile Shinysdr to activate the mods?
cheers!

kpreid, 2018-02-16:
> Is there a way to modify the VOR plugin to receive NOAA APT weather satellites?(or a other plugin)

It would make more sense to write a new plugin. The VOR plugin is not particularly suited as a starting point. If there is an available program to decode APT signals delivered over a pipe (I have not looked at the topic) then `shinysdr/plugins/multimon.py` would be most suitable to look at as a starting point — it uses a FM demodulator followed by an external process to decode APRS telemetry.

Displaying the image data on the client will also require some work, but it makes sense to do it since the same techniques could work for e.g. SSTV. I will take this issue as a suggestion to work on that.

> I need to modify the filter bandwidth to 36Khz for the APT signal.

This is no problem. Each demodulator plugin does its own filtering.

> so if the VOR plugin has been modified, must we recompile Shinysdr to activate the mods?

There is no compilation. If you followed [the setup.py style installation instructions \(4a\)](https://kpreid.github.io/shinysdr/manual/installation) then you must rerun the installation in order to get the changes to take effect; if you are running from source (4b) then nothing is necessary beyond the code changes.

AlanJan, 2018-02-25:
Hi Kpreid,

Thank You for your reply!

For the image decode section of the weather satellites, I use WXtoImg and for audio routing from ShinySDR(Chrome) to WXtoImg (within windows) I use VBCABLE.
But now I need a FM demodulator with a bandwidth of 36 Khz.
or a demodulator with a adjustable demodulator bandwidth will be fine.

I have look at the multimon.py but i can't make any sense of it.
Do you have some kind of sample / basic demodulator that i can use for testing?

I have compare ShinySDR with other programs like OpenWebRX for my need and I must say, I like your  ShinySDR very mutch!, I have installed ShinySDR permanent on my server and with my VPN connection i can use it every where.
Now,  with a APT plugin, it will be the perfect communication wide band receiver by software!
Well done!

Cheers!
Alan

kpreid, 2018-02-25:
Thank you for clarifying your needs. If you wish to experiment with changing the bandwidth yourself, the relevant code is in `shinysdr/plugins/basic_demod.py` as `NFMDemodulator`. The filter bandwidth is defined in terms of the expected FM deviation, but that is not necessary. To create a new mode option you would need to copy both `NFMDemodulator` and `pluginDef_nfm` which refers to it (and give the copy a new mode ID).

I am working on some steps towards having an actual APT plugin, but this may or may not happen any time soon. (Adding the ability to simply choose a different filter from the UI is another possible.)

AlanJan, 2018-02-26:
Hi,

> If you wish to experiment with changing the bandwidth yourself, the relevant code is in shinysdr/plugins/basic_demod.py as NFMDemodulator. The filter bandwidth is defined in terms of the expected FM [...]

Thank you for your tip!. I have made a raw plugin and the first test works great, but i have set the bandwidth to 15000 (deviation = 15000), so now waiting for the next satellite to come over;-) 

cheers!