kpreid/shinysdr #108: Scrolling not working on Firefox

cameronswinoga, 2018-03-06:
First off, absolutely love this project!  Having a blast using my SDR on my server while away from home. 

That said, I've noticed that when I use Firefox to access the web interface I cannot use the scrolling features i.e. zooming the waveform, mouse scrolling on the numbers, etc.

cameronswinoga, 2018-03-06:
Also as a side note, I cannot figure out what the 'Map' and 'Telemetry' panels are supposed to do or how I'm supposed to use them.  As a side side note, is there log output for what external plugins are/are not being loaded?

kpreid, 2018-03-06:
> when I use Firefox to access the web interface I cannot use the scrolling features i.e. zooming the waveform, mouse scrolling on the numbers, etc.

Spectrum zooming on Firefox [is currently disabled](https://github.com/kpreid/shinysdr/blob/b7ee33da5e68b244088d7e485d669ff8a8122bd6/shinysdr/i/webstatic/client/widgets/spectrum.js#L106) because of a page layout difference/bug between Blink/WebKit and Gecko which would produce a much more unusable mess.

The number control has no reason not to work — except that I see that I forgot to fix the event handler when I was working on Firefox compatibility. Fixed in commit 53f1281fe9c14513b8829ac27ef92d1d30154183. Thanks for noticing!

> I cannot figure out what the 'Map' and 'Telemetry' panels are supposed to do or how I'm supposed to use them.

Added to the 'needs documentation' list. Though I can see also having an explanation in-page when Telemetry is blank.

The Map pane (which should not be blank but displaying a map/globe with at least continent borders) is for displaying location information which can be either fixed (the locations of transmitters recorded in the frequency database) or live (the locations reported over APRS or ADS-B protocols).

The Telemetry panel also shows that live location information as well as non-location information decoded from APRS, ADS-B, or the miscellany that rtl_433 supports. All you need to do is select one of those demodulators and wait for a packet to be received and the information will show up in Telemetry and Map as appropriate.

> As a side side note, is there log output for what external plugins are/are not being loaded?

If a plugin actually crashes, it will show up in the stderr of the server. If it merely decides to declare itself unavailable, there is no logging yet, but I am working on having a "server status" page which includes explanations of unavailability as well as other read-only administrative information like counting connected clients.

cameronswinoga, 2018-03-06:
The reason I ask about the plugin logging is because I think I installed all of the optional plugins correctly, however rtl_443 is the only additional demodulation I have available

kpreid, 2018-03-07:
Commit b68951b93787f289cf30beed8e4a44e334273bac adds logging of reasons for unavailability. Look for `Loading mode plugins...` followed by the errors if there are any.

kpreid, 2018-03-12:
I believe this has all been addressed except for the documentation, and I've filed another issue for that.

closed 2018-03-12 by kpreid