kpreid/shinysdr #112: ValueError: Must have at least one RF device

quirq, 2018-04-08:
When trying to run shinysdr I get this error message: 

```
INFO:shinysdr:Log opened.
WARNING:shinysdr:/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/config.py:79: exceptions.UserWarning: No network service defined!
INFO:shinysdr:Constructing...
CRITICAL:shinysdr:main function encountered error
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 52, in main
    react(go)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-18.4.0rc1-py2.7-linux-x86_64.egg/twisted/internet/task.py", line 908, in react
    finished = main(_reactor, *argv)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 47, in go
    return _main_async(reactor, argv, _abort_for_test)
  File "/usr/local/lib/python2.7/dist-packages/Twisted-18.4.0rc1-py2.7-linux-x86_64.egg/twisted/internet/defer.py", line 1532, in unwindGenerator
    return _inlineCallbacks(None, gen, Deferred())
--- <exception caught here> ---
  File "/usr/local/lib/python2.7/dist-packages/Twisted-18.4.0rc1-py2.7-linux-x86_64.egg/twisted/internet/defer.py", line 1386, in _inlineCallbacks
    result = g.send(result)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 98, in _main_async
    app = config_obj._create_app()
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/config.py", line 88, in _create_app
    features=self.features._get_all())
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/session.py", line 37, in __init__
    features=features)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/top.py", line 70, in __init__
    raise ValueError('Must have at least one RF device')
exceptions.ValueError: Must have at least one RF device

INFO:shinysdr:Main loop terminated.
```

I have my device attached and had it working with Gqrx, so that should be fine.
Any other help? :)

quirq, 2018-04-08:
Tried reinstalling everything, leaving defaults and blocking rtl-sdr from stealing control of the stick.
All is now working!

closed 2018-04-08 by quirq

KC9S, 2019-06-24:
I know this thread is long closed but how did you block rtl_sdr?  I have three sticks and only want shinysdr to control one of them.  Thanks.