kpreid/shinysdr #113: ACARS

k9wkj, 2018-04-11:
I found https://github.com/antoinet/gr-acars2 and wondered if it could be made to pull in on setup
de k9wkj

k9wkj, 2018-04-11:
okay it seems the ADSB block does decode the packets but parsing is ahhh  shall we say odd

(-33 207.80955600) No handler for message type 24 from a94810
(-32 208.84759150) Type 5 (short surveillance ident reply) from 6dcb50 with ident 1306 (SPI)
(-34 209.74119850) Type 4 (short surveillance altitude reply) from 50bd10 at 82200ft (GROUND ALERT)
(-32 210.11998850) Type 5 (short surveillance ident reply) from 3728b0 with ident 126 (AIRBORNE ALERT)
(-34 210.26432900) No handler for message type 24 from 67a471
(-33 210.35154650) No handler for message type 24 from 6dd55a
(-34 210.56155900) Type 5 (short surveillance ident reply) from 1197fa with ident 356 (aircraft is on the ground)
(-31 210.71629100) No handler for message type 24 from 46ee1d
(-33 212.04815400) No handler for message type 24 from 509266
(-33 212.19612900) Type 5 (short surveillance ident reply) from 755ea2 with ident 4220 (aircraft is on the ground)
(-34 212.58121900) Type 0 (short A-A surveillance) from 4fa8da at 33325ft (Full TCAS resolution) (aircraft is on the ground)
(-34 328.40421800) Type 5 (short surveillance ident reply) from 665866 with ident 3452 (SPI ALERT)
(-35 328.51816100) Type 0 (short A-A surveillance) from 5cf434 at 119200ft (Full TCAS resolution) (aircraft is on the ground)
(-34 328.53158450) Type 5 (short surveillance ident reply) from db0a71 with ident 1724 (aircraft is on the ground)
(-33 328.97364900) Type 0 (short A-A surveillance) from b955ef at 60100ft (TCAS resolution inhibited)
(-33 329.09921650) Type 5 (short surveillance ident reply) from e26aec with ident 6240 (SPI ALERT)
(-35 330.49266150) No handler for message type 24 from 42cb94
(-32 331.82584700) No handler for message type 24 from b05cee
(-33 332.06323750) No handler for message type 24 from 8c5f46
(-34 332.31176100) Type 5 (short surveillance ident reply) from 2ae3f7 with ident 7302 (AIRBORNE ALERT)
(-32 332.58535050) No handler for message type 24 from c78a90

kpreid, 2018-04-12:
FYI, some notes on the situation:

* All of the text you have quoted comes from gr-air-modes and isn't under ShinySDR's control — that is, I can't improve the decoding if there's a problem. (Also, there is some junk, depending on how low you set the *Decode threshold* control.)
* There's a lot of information that is available but not displayed by ShinySDR, because I don't know enough about the details of these protocols to get the information out of gr-air-modes's returned data structures and label it correctly. Contributions would be appreciated (the relevant code is in https://github.com/kpreid/shinysdr/blob/master/shinysdr/plugins/mode_s/__init__.py, class `Aircraft`).

I haven't had a chance to look at gr-acars2 yet but I am interested in the idea.

k9wkj, 2018-04-12:
ohh yes I know thats coming from gr-air-modes
but i thought it was interesting that it was decoding the packets but couldnt parse things very well
i know that ACARS is not the same protocol as ADSB
and i didnt see anything in the docs of gr-air-modes that would suggest it could handle ACARS