kpreid/shinysdr #125: No audio on initial launch

ofadam, 2019-01-23:
When I initially launch ShinySDR via any web browser, the audio is muted although the waterfall display is active and showing signals. I must first click on the "Direct link to audio stream" first, then go back to the dashboard to have working audio. Any idea what may be causing this?

kpreid, 2019-01-24:
Which browsers, exactly? In Chrome, I would guess this is due to the newly enabled [autoplay restrictions](https://developers.google.com/web/updates/2017/09/autoplay-policy-changes), but that shouldn't trigger on repeated visits to the page.

Is it really the "direct link" that makes it work, or is it just navigating away and back? (Try the other sidebar links, like "ShinySDR Manual".)

ofadam, 2019-01-24:
I'm using the latest version of Chrome for OSX, but it also is happening on
Safari.

I tried using other sidebar links and it doesn't free up the audio like
clicking on that "direct link."

ofadam, 2019-01-24:
[image]

There's a strange sawtooth shape present when no audio.

ofadam, 2019-01-24:
Firefox works, though!

kpreid, 2019-01-24:
That sounds like the AudioContext isn't running (the sawtooth is the audio buffer repeatedly filling up and being thrown out). Please check the console for messages.

kpreid, 2019-01-24:
Sorry, I mean the browser console.

ofadam, 2019-01-24:
Ah, right.

"The AudioContext was not allowed to start. It must be resumed (or created) after a user gesture on the page."

kpreid, 2019-01-24:
In my experience, that message can appear even if the AudioContext is working normally. Could you please copy the _entire_ console text so I can see the sequence of messages?

(I'll probably need to add some extra debugging to figure this out, but the complete log might have hints.)

ofadam, 2019-01-24:
How's this?

main.js:95 Loading plugins…
main.js:95 Connecting to server…
ws-stream.js:68 The AudioContext was not allowed to start. It must be resumed (or created) after a user gesture on the page. https://goo.gl/7K7WLu
connectAudio @ ws-stream.js:68
connectRadio @ main.js:166
plugins @ main.js:113
execCb @ require.js:1690
check @ require.js:865
(anonymous) @ require.js:1140
(anonymous) @ require.js:131
(anonymous) @ require.js:1190
each @ require.js:56
emit @ require.js:1189
check @ require.js:940
(anonymous) @ require.js:1140
(anonymous) @ require.js:131
(anonymous) @ require.js:1190
each @ require.js:56
emit @ require.js:1189
check @ require.js:940
enable @ require.js:1177
init @ require.js:783
(anonymous) @ require.js:1015
(anonymous) @ require.js:131
finishLoad @ text.js:152
(anonymous) @ text.js:188
xhr.onreadystatechange @ text.js:300
XMLHttpRequest.send (async)
text.get @ text.js:308
load @ text.js:187
(anonymous) @ require.js:1096
(anonymous) @ require.js:131
on @ require.js:512
callPlugin @ require.js:957
fetch @ require.js:819
check @ require.js:851
enable @ require.js:1177
enable @ require.js:1550
(anonymous) @ require.js:1162
(anonymous) @ require.js:131
each @ require.js:56
enable @ require.js:1114
init @ require.js:783
callGetModule @ require.js:1204
completeLoad @ require.js:1583
onScriptLoad @ require.js:1711
load (async)
req.load @ require.js:1939
load @ require.js:1679
load @ require.js:829
fetch @ require.js:819
check @ require.js:851
enable @ require.js:1177
enable @ require.js:1550
(anonymous) @ require.js:1162
(anonymous) @ require.js:131
each @ require.js:56
enable @ require.js:1114
init @ require.js:783
(anonymous) @ require.js:1453
setTimeout (async)
req.nextTick @ require.js:1809
localRequire @ require.js:1442
requirejs @ require.js:1791
main @ main.js:112
(anonymous) @ (index):101
execCb @ require.js:1690
check @ require.js:865
[... 216 lines cut ...]
createWidgets @ widget.js:326
(anonymous) @ widget.js:127
createWidgetsInNode @ widget.js:126
createWidgets @ widget.js:357
(anonymous) @ widget.js:127
createWidgetsInNode @ widget.js:126
go @ widget.js:238
startNow @ events.js:67
createWidget @ widget.js:149
createWidgetExt @ widget.js:274
PaneImpl @ pane-manager.js:130
container.querySelectorAll.forEach.element @ pane-manager.js:297
_addExisting @ pane-manager.js:292
PaneManager @ pane-manager.js:243
initialStateReady @ main.js:218
_RAFCallback @ events.js:147
requestAnimationFrame (async)
enqueue @ events.js:117
nSchedule @ events.js:219
Notifier.notify @ events.js:225
ReadCell._update.data @ network.js:142
updaterMap.(anonymous function) @ network.js:367
oneMessage @ network.js:403
ws.onmessage @ network.js:457
main.js:95 Ready.

kpreid, 2019-01-24:
OK, unfortunately, that's essentially the same set of messages I see routinely. I've got some ideas, but it'll probably be a few days before I have them out for you to test.

ofadam, 2019-01-24:
Understood! I really appreciate all the attention you've give it already. It's been an impressive piece of software.

kpreid, 2019-01-25:
I pushed 446e6dd45ffbbebf82812e2d4bb2b521984d5746 which adds debug logging.

For me, on reloading, I get this for a log:

```
main.js:95 Loading plugins…
main.js:95 Connecting to server…
ws-stream.js:72 The AudioContext was not allowed to start. It must be resumed (or created) after a user gesture on the page. https://goo.gl/7K7WLu
ws-stream.js:68 audio playback debug: initial state is suspended
main.js:95 Downloading state…
widget.js:225 Widget Banner did not handle title attribute
main.js:95 Ready.
[Violation] 'requestAnimationFrame' handler took 405ms
ws-stream.js:257 Streaming using AudioWorklet 44100 2ch audio and converting to 44100
ws-stream.js:68 audio playback debug: attempting resume, immediate state is suspended
ws-stream.js:283 The AudioContext was not allowed to start. It must be resumed (or created) after a user gesture on the page. https://goo.gl/7K7WLu
ws-stream.js:68 audio playback debug: attempted resume, immediate state is suspended
ws-stream.js:68 audio playback debug: resume resolved
[Violation] 'requestAnimationFrame' handler took 302ms
```

So the Chrome autoplay policy seems to be not doing what it's supposed to do, but for me the final result is that playback works anyway. Let me know what your log looks like. No need to copy the stack traces.

ofadam, 2019-01-25:
> I also notice in your log that the audio stream is running in
> ScriptProcessor mode rather than AudioWorklet mode. This is surprising.
> Could you specify exactly which Chrome version you're using?

I'm using 71.0.3578.98 (64-bit) for OS X.

kpreid, 2019-01-26:
@ofadam Please rerun with the latest ShinySDR revision and report the new log results. No need to include the stack traces.

ofadam, 2019-05-13:
It's been a while since I worked on this, but the issue remains even in the latest release. Starting the webapp results in silence with a sawtooth audio wave, which is only fixed after clicking on that direct audio link.

kpreid, 2019-05-21:
I've been trying to make progress towards a robust fix for this.

Ignoring the question of any issues with Chrome's actual implementation of the stated autoplay policy, in order to make this work in general we definitely need a client-side state of “muted / not playing” which can then be “unmuted” by the user. I'm making some changes to the UI layout to make a good place for that mute button.

mathisono, 2019-05-21:
It would be nice to enable the audio stream for a specific recever. Say i
dont audio in the browser, but just on the stream? Then open up a second
recever and hear that in the browser.

I notced a few bug... But i will wait untill you have pushed the update to
report on bugs.

Mathison

kpreid, 2019-05-21:
Having audio streams with different content is a bigger problem — right now, the server-side signal flow is completely shared, the only exception being the sample rate conversion for audio. Changing that is covered by issue #73.

And if you've observed bugs, please do report them as separate issues. ShinySDR development is mainly limited by having enough free time to think about how to refactor things, and user-visible bugs are often really simple to tackle so it'll actually get things done faster in total if I know about more of them.

kpreid, 2019-07-08:
I was doing some refactoring to prepare to have the above-mentioned mute state and I discovered that it wouldn't actually be guaranteed to resume the context on startup if there's supposed to be audio playing. Oops. Commit aeaaabe0f5e16fc77b62ccc105ded0cefe01ac50 fixes that, I hope.

@ofadam Give this update try and let me know if it does anything different?

ofadam, 2020-01-29:
Hello, I'm back! And still having trouble with audio not playing on initial launch with Chrome. It does with Firefox, somehow. Any further suggestions for ironing this out? I get the sawtooth-like waveform in the audio panel.

kpreid, 2020-01-31:
@ofadam Please update ShinySDR to the latest version and post the contents of your console on one of the loads that fails to play audio (no stack traces needed, just the log messages for now).

ofadam, 2020-01-31:
`Loading plugins…
main.js:95 Connecting to server…
ws-stream.js:72 The AudioContext was not allowed to start. It must be resumed (or created) after a user gesture on the page. https://goo.gl/7K7WLu
connectAudio @ ws-stream.js:72
connectRadio @ main.js:166
(anonymous) @ main.js:113
execCb @ require.js:1690
check @ require.js:865
(anonymous) @ require.js:1140
(anonymous) @ require.js:131
(anonymous) @ require.js:1190
each @ require.js:56
emit @ require.js:1189
check @ require.js:940
(anonymous) @ require.js:1140
(anonymous) @ require.js:131
(anonymous) @ require.js:1190
each @ require.js:56
emit @ require.js:1189
check @ require.js:940
enable @ require.js:1177
init @ require.js:783
(anonymous) @ require.js:1015
(anonymous) @ require.js:131
finishLoad @ text.js:152
(anonymous) @ text.js:188
xhr.onreadystatechange @ text.js:300
XMLHttpRequest.send (async)
text.get @ text.js:308
load @ text.js:187
(anonymous) @ require.js:1096
(anonymous) @ require.js:131
on @ require.js:512
callPlugin @ require.js:957
fetch @ require.js:819
check @ require.js:851
enable @ require.js:1177
enable @ require.js:1550
(anonymous) @ require.js:1162
(anonymous) @ require.js:131
each @ require.js:56
enable @ require.js:1114
init @ require.js:783
callGetModule @ require.js:1204
completeLoad @ require.js:1583
onScriptLoad @ require.js:1711
load (async)
req.load @ require.js:1939
load @ require.js:1679
load @ require.js:829
fetch @ require.js:819
check @ require.js:851
enable @ require.js:1177
enable @ require.js:1550
(anonymous) @ require.js:1162
(anonymous) @ require.js:131
each @ require.js:56
enable @ require.js:1114
init @ require.js:783
(anonymous) @ require.js:1453
setTimeout (async)
req.nextTick @ require.js:1809
localRequire @ require.js:1442
requirejs @ require.js:1791
main @ main.js:112
(anonymous) @ (index):99
execCb @ require.js:1690
check @ require.js:865
(anonymous) @ require.js:1140
(anonymous) @ require.js:131
[... 123 lines cut ...]
startNow @ events.js:67
createWidget @ widget.js:155
createWidgetExt @ widget.js:285
PaneImpl @ pane-manager.js:138
(anonymous) @ pane-manager.js:311
_addExisting @ pane-manager.js:306
PaneManager @ pane-manager.js:259
initialStateReady @ main.js:218
_RAFCallback @ events.js:147
requestAnimationFrame (async)
enqueue @ events.js:117
nSchedule @ events.js:219
Notifier.notify @ events.js:225
ReadCell._update @ network.js:142
updaterMap.<computed> @ network.js:367
oneMessage @ network.js:403
ws.onmessage @ network.js:457
main.js:95 Ready.
[Violation] 'requestAnimationFrame' handler took 491ms
[Violation] Forced reflow while executing JavaScript took 112ms
[Violation] 'requestAnimationFrame' handler took 60ms
[Violation] Forced reflow while executing JavaScript took 59ms
[Violation] Forced reflow while executing JavaScript took 39ms
bufferer.js:106 Extreme audio overrun.
[Violation] 'requestAnimationFrame' handler took 277ms`

kpreid, 2020-02-02:
[I've edited your comment to strip out the stack traces which make it hard to read.]

I don't have any good ideas about why this is happening, but apparently the autoplay policy isn't behaving the same for you as others. (Do you routinely do a 'clear browsing data', or use some other sort of privacy tool, maybe?)

The correct solution here is for ShinySDR to implement a UI to prompt for starting playback when it is needed (as the "audio toolbox" pages already do). I don't seem to have it recorded as an issue already, but the plan I had in mind was to add a “mute/pause” feature to the audio UI. Then, if autoplay is not permitted (or, perhaps, if the user prefers it), the page will start in muted state and the user will have to click to unmute it, which should satisfy the autoplay policy without extra dialogs.

redge76, 2020-11-27:
Temporary and local solution: https://github.com/kpreid/shinysdr/issues/148#issuecomment-734713774