kpreid/shinysdr #126: No module named i.db ................ On a fresh install

mathisono, 2019-02-08:
Am I missing something here or is it the code?

mat@radcom:~/shinysdr$ shinysdr
Traceback (most recent call last):
  File "/usr/local/bin/shinysdr", line 11, in <module>
    load_entry_point('ShinySDR==0.0.0', 'console_scripts', 'shinysdr')()
  File "/home/mat/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/mat/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/home/mat/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/home/mat/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/mat/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 36, in <module>
    from shinysdr.config import Config, write_default_config, execute_config
  File "/home/mat/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/config.py", line 39, in <module>
    from shinysdr.i.db import DatabaseModel, database_from_csv, databases_from_directory
ImportError: No module named i.db

kpreid, 2019-02-08:
Should be fixed with 112417bc8a30569a3b6169e7ee22faf1df7aa21f. Please let me know.

mathisono, 2019-02-08:
That took some work!!! 
The current install instructions slipped me up I had to fight to get the program uninstalled !!!
I suggest the following  for 4a.

# system-wide installation
cd shinysdr
./fetch-js-deps.sh
python setup.py build
sudo pip install ./

WORKS!

closed 2019-02-09 by kpreid