kpreid/shinysdr #127: Frequency database doesn't always load fully

kpreid, 2019-02-10:
Sometimes some frequency databases' content is missing in the spectrum view, and reloading helps.

I haven't dug into this yet, but it probably has to do with JS database objects' change notifications.

mathisono, 2019-02-12:
I noticed that a reload clears it up!

 But only to find out that new database entrys are not retained after i kill the program. After adding a few entrys for aprs and ATC, they are gone!? I havent chased down the bug.  when we add a new entry in the web ui,  is it amending one of the db or starting a new one?

kpreid, 2019-02-12:
@mathisono To get entries saved, you need to `config.databases.add_writable_database` as described in the manual. However, you made me notice that this is a terrible new user experience, so I'll look into fixing that (#129).